ArduPilot/ardupilot #7435: Climb rate and Thr.out issue when switching from RLT to Loiter

SergeyBokhantsev, 2017-12-20:
#### Issue details
Steps:
1. Initialize RTL. Copter climbs to RTL heigth, moves to launch location and begins descending.
2. At RTL-descending phase switch to Loiter

Result: I can hear motors cut-off for a moment (CTUN.ThO = 0 in log), then copter stabilizes.

Two issues here:
- Why CTUN desired climb rate goes down rapidly after switching to Loiter
- Why CTUN throttle out goes zero when my THR_MIN > 0?

Some additional info here: https://discuss.ardupilot.org/t/throttleout-is-zero-for-a-moment-when-switched-from-rtl-to-loiter/22663/3

#### Version
3.5.3

#### Platform
[ x] Copter

#### Airframe type
X

#### Hardware type
Pixracer

#### Logs
https://www.dropbox.com/s/zfacgo25mcm448i/00000022.BIN?dl=0

lthall, 2018-01-11:
I understand what is happening here. This will be fixed.

SergeyBokhantsev, 2018-05-31:
hi @lthall, what is the current status of this?

rmackay9, 2018-10-25:
I think this issue is fixed in master and Copter-3.6.  Below is a SITL test in which I switched the vehicle into Loiter mode a couple of times as it was descending in RTL.  No throttle spike as far as I can see.

[image]

rmackay9, 2018-10-25:
.. although it's possible that the way I'm testing is the problem.  From looking at the logs from @SergeyBokhantsev above, it's clear the desired climb rate was not being initialised correctly.
[image]

I'll check with @lthall if he remembers the status of this.  @SergeyBokhantsev if you could try to reproduce this with Copter-3.6 that would also be great.  I think we should continue this discussion in the support forums so I've created a topic [here](https://discuss.ardupilot.org/t/throttle-drop-when-switching-to-loiter-from-rtl/34388).

SergeyBokhantsev, 2018-11-14:
@rmackay9 
Test to repro: https://www.dropbox.com/s/tne9o1r46nogoia/test.txt?dl=0
Produces the same issue with SITL:
[image]

Also I've run same test with [this condition](https://github.com/ArduPilot/ardupilot/blob/e702ca7202b8deaf3141a3829f7377b7d9396ab4/ArduCopter/mode_loiter.cpp#L28) removed (but not the condition body).
This change does fix the issue 
[image]

thought I'm not sure why that condition here and if this is correct way to fix. Another concern is that the same pieces of code are in another modes init().

lthall, 2021-09-27:
All this code has been rewritten in 4.1 with more stringent initialisation.

closed 2021-09-27 by lthall