ArduPilot/ardupilot #7437: pixhawk 2.4.6. PX4 flight stack no magnetometer

FlyingBeginner, 2017-12-20:
#### Issue details
Hi @ all,
I am very new to the topic, so may be there exists a very simple solution - which would be great at all.
I uploaded the latest px4 flight stack 1.7.0 on my pixhawk 2.4.6 and cannot arm the quadrocopter. There is an error stating "no mag sensor 0".
I can exclude a hardware error, as the magnetometer works fine with ardupilot.
Does anyone know about the issue or has some suggestions?
Thank you so much!
Best regards,
Jan

#### Version
px4 flight stack 1.7.0

#### Platform
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
X

#### Hardware type
pixhawk 2.4.6 no external sensors

#### Logs
_Please provide a link to any relevant logs that show the issue_

auturgy, 2017-12-20:
Load arducopter firmware and see if the problem persists 

Regards,

James

lvale, 2017-12-20:
Unrelated issue to ArduPilot

closed 2017-12-20 by lvale