ArduPilot/ardupilot #7439: External ICM20948 Compass on HERE Module not detected

caomuqing, 2017-12-21:
#### Issue details
I am reporting a very strange issue we have about our new HERE GNSS with ICM20948 compass. When the new HERE modules are connected with pixhawk2.1 flight controllers, some of the external ICM20948 compass cannot be detected with the new 3.5.4 firmware, or new 3.8.3 firmware. However, on some pixhawk2.1 hardware, it is detected without issue.

To rule out hardware issue, I compiled and uploaded a earlier pull request by Mr Tridge here: https://github.com/ArduPilot/ardupilot/pull/7030 , using this firmware, all pixhawk2.1s are able to detect the external ICM20948 compass.

Is there a possibility that the new compass detections on boot fail to detect some compass?

#### Version
Plane 3.8.3
Copter 3.5.4

#### Platform
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type

#### Hardware type
HERE with ICM20948, Pixhawk2.1
#### Logs
This is the log when ICM is detected:
[17 1-1-2000 8-28-34 AM.bin.log](https://github.com/ArduPilot/ardupilot/files/1577712/17.1-1-2000.8-28-34.AM.bin.log)
This is the log when ICM is not detected:
[47 2000-1-1 8-11-02.bin.log](https://github.com/ArduPilot/ardupilot/files/1577714/47.2000-1-1.8-11-02.bin.log)
Both uses pixhawk2.1 and HERE module

proficnc, 2017-12-21:
Along side this issue is the incorrect orientation of the sensor in the driver.

expntly, 2018-03-08:
I can confirm this issue.  However, my findings are the following.  ArduPlane (3.8.x) won't detect the device, but ArduCopter (3.5.5) will detect it just fine.  I've tested it with half a dozen Here GNSS with ICM20948, so this is not an isolated case.  Somehow the HMC5883 chip seems to work better.

COMPASS_TYPEMASK=32

rmackay9, 2018-03-08:
I think @tridge is planning a plane release in the next month or so.  I think he's just been very busy with the ChibiOS project but that is nearing completion I think... at least the major part of the project is nearing completion.

proficnc, 2018-03-08:
The hardware is fine, just waiting on a new plane release.
Copter and Rover are all caught up.

Plane Master is also caught up.

mwrightE38, 2018-03-11:
@rmackay9 @proficnc  I've confirmed that this is working just fine on pixhawk 2.1 with Copter V3.5.5  as well.

However with the same firmware compiled for pixhawk the compass isn't detected. Is there a known reason for this?

WickedShell, 2018-06-25:
@caomuqing @proficnc can this be closed? Plane has started it's release cycle so I think the relevant fixes should be shipping everywhere soon(ish).

proficnc, 2018-06-25:
All good, let’s close

caomuqing, 2018-08-26:
This issue is not completely resolved: we have some users reporting that the new ICM compass on HERE cannot be detected on some pixhawk1 (and the cube) hardware. We have done some testing and found out that:
This is not HERE hardware issue. The issue is repeatable on flight controller with different pieces of HERE hardware.
This is partially a autopilot hardware issue: on most of the pixhawk1 and cube hardwares the ICM compass can be detected, but on some of them it cannot be detected.
However, this is also a software issue: for those hardwares that cannot detect ICM20948, if we change the firmware from the latest firmware to the one in  #7030 , the ICM compass can be detected without issue.
The root of the problem remains unknown, my guess is that each piece of hardware has very slight difference when powered up and the sensor data comes in at slightly different time. However, with the recent changes on compass detection scheme (maybe the addition of COMPASS_TYPEMASK), this difference is significant enough for some compass to be not detected on power up.

Any suggestion is greatly appreciated!

rmackay9, 2018-08-27:
@caomuqing, I think it might be because of the difference between -v2 and -v3 firmware.  I've seen the MP detect a board that needs the -v2 firmware when in fact the -v3 will fit.  I think the compass doesn't work with the -v2 firmware.

caomuqing, 2018-08-27:
@rmackay9 thanks for your insight! The could be the cause. In that case, is there a way to make sure the correct firmware version to be loaded?