ArduPilot/ardupilot #7444: Aurduplane stopping the pusher motor before a load drop

JeyPi, 2017-12-22:
Feature request:
Is there a way of stopping the electric motor during an auto flight?
The reason of that feature is to stop the propeller revolutions just before a load is released from a pusher plane. That way the propeller would be safe in case the load is light and gets blown in to the prop disk upon release. This is my problem at the moment as it happens  8 times out of 10 flights with my airframe.
I emagine it should work in a similar way as a chute release.

#### Platform
Plane

#### Airframe type
Pusher

geofrancis, 2017-12-27:
would do_set _servo work with the throttle channel?

magicrub, 2017-12-27:
related to https://github.com/ArduPilot/ardupilot/issues/890 (stop motor for picture)

JeyPi, 2018-01-01:
Thanks. I went through that stop motor for picture before. I thought it could be related or at least use the same procedure in flight. My pusher has a folding prop. When the motor uses a break it stops immediately within 0.5 sec.