ArduPilot/ardupilot #7461: Add "breadcrumb" Rover mode

peterbarker, 2017-12-27:
#### Issue details

Copter's chase-mode (PR) shows how a flight mode can interact with other vehicles.

A mash-up of that PR and SmartRTL would allow a vehicle to follow a vehicle using a breadcrumb trail.

That is to say, monitor the other vehicle's `GLOBAL_POSITION_INT` mavlink message.  Use something akin to SmartRTL's breadcrumb storage to store the points (without the loop-detection/elimination).  Follow the breadcrumb trail at either a fixed rate or to maintain a following distance to the first vehicle.

#### Platform
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[ X ] Rover
[ X ] Submarine

peterbarker, 2017-12-27:
```
Philip, 11:09
Imagine it following a skier around obstacles... but keeping the gimbal trained on the skier... wow! That would be incredible! 
```

Note "vehicle" here could mean "GCS" (e.g. a mobile phone).

rmackay9, 2021-08-17:
I think I might close this if that's OK.  It's an interesting idea but I don't think a user has actually asked for this yet.  If a user does appear asking for this we can re-open it.

closed 2021-08-17 by rmackay9