ArduPilot/ardupilot #7463: Arduplane 3.8.3 : Servos not working (digital)

DroneExplorer, 2017-12-27:
#### Issue details
Hi, 

After installing Arduplane 3.8.3, Servos stopped working.
After disconnecting 2 out of 4, they work but they don't follow the commmand sent buy the radio and do not bounce when centered. Doesn't depend on which 2 servos are disconnected.

Went back to 3.7.1 and 3.8.2 and it works perfectly.

Regards,
Hamza

#### Version
Arduplane 3.8.3

#### Platform
[  ] Pixhawk 2.1
[  ] Plane

#### Airframe type
VTAIL

#### Hardware type
Pixhawk 2.1

#### Logs
Not Relevant here.

magicrub, 2017-12-27:
updated to v3.8.3 from what verison?

DroneExplorer, 2017-12-27:
Fresh install, on a new Pixhawk 2.1

magicrub, 2017-12-27:
oh, didn't see that you said 3.7.1 before. My mistake.

DavidIngraham, 2018-01-12:
What is the model of servo that you are using? Also, could you provide a log or param file? Additionally, how are you providing power to the servo rail?

svwhisper, 2018-01-17:
Same issue here with a fresh download of 3.8.4.  I don't know the brand of servo, as they are pre-installed on an Opterra wing.  The Pixhawk 2.1 throttle output works ok but aileron servos are dead.  Directly connecting the servos to the Frsky receiver works fine, with the servos operating as expected using Taranis RC.  Monitoring the mavlink messages emitted by the Pixhawk as I cycle the Taranis stick, I can see the values of the aileron channels changing as I would expect, but the servos are inoperative.

I would dearly love to know how to downgrade to 3.8.2 to see if that resolves the issue.  Can anyone point me to a downloadable binary?

DroneExplorer, 2018-01-17:
@DavidIngraham I am using corona DS-929MG which work fine on 3.8.2 and 3.8.4, still nothing on 3.8.3.
Servo rail is powered through a dedicated 5v bec (castle bec).
SOrry, i am in the process of moving and all my stuff is packed I don't have access to the plane to give you log and param files.

DroneExplorer, 2018-01-17:
@svwhisper For me it worked on 3.8.4, my problem was with 3.8.3, but this may depend on the brand and type of servo (digital/analog).

To downgrade you just have to install rover for example, then reinstall plane 3.8.2 by clicking on previous firmware download on mission planner

svwhisper, 2018-01-18:
I attempted without success to downgrade firmware.  However, it would seem the process of installing rover and then plane 3.8.4 has resolved my issue.  The servos now work as they should.  Apologies.

closed 2018-01-18 by magicrub