ArduPilot/ardupilot #7465: How to change px4 microprocessor

afeizinejad, 2017-12-28:
#### Issue details
Hello.
I'm going to change the Pixhawk processor from stm32f4 to stm32f7. How do I change the code to be compatible with the new processor?
Any help would be appreciated.
Thanks.

#### Version
ArduCopter V3.0.1

#### Platform

[  ] Copter

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
pixhawk

#### Logs

magicrub, 2017-12-28:
already done. 

make px4-v4

auturgy, 2017-12-28:
V4 is just another f4 target.
The v5 is an f7, but I think there are only 20 of them, and no ArduPilot build target yet.

Regards,

James

DavidIngraham, 2018-01-12:
Hello @afeizinejad, Support for F7 processors is being added through the ChibiOS port. I encourage you to look at the ArduPilot Chibios gitter channel for support: https://gitter.im/ArduPilot/ChibiOS

The chibios team is creating tools that make it trivial to port ArduPilot to new STM32 processors. That effort is well tracked, so I think we can close this issue now.

closed 2018-01-12 by DavidIngraham