ArduPilot/ardupilot #7469: ArduPlane support for Waypoint Injections in AUTO mode

oneTimePad, 2017-12-30:
#### Issue details
My team and I are working on a UAV path-planning project with ArduPilot.
We would like to be able to plan a mission prior to flight, and then as the aircraft sees obstacles, a path-planning algorithm would inject waypoints via a MAVLink message to modify the current mission to avoid obstacles.

I see there is guided mode, but there doesn't seem to be a mode that supports prior mission planning and then in-flight mission modification.

I forked this repo and made the following modifications to allow for this (I worked on ArduPlane-release, for now, since we will be using it with the Pixhawk and, for our tests, want a stable version.):

The mission proceeds as planned until a MAV_CMD_INJECT_WAYPOINT is sent. When received, this command is added to the end of the current mission. ArduPilot then suspends the current active nav command and proceeds to the injected waypoint. Injected waypoints are searched for past the length of the original mission in memory. Once all injected waypoints are exhausted, ArduPilot resumes the nav command that was interrupted.

It seems to work in SITL so far. My team and I plan eventually to test it with the Pixhawk.

I was wondering if this might be a useful feature to integrate into ArduPilot? If, I can link modifications.

Thank you

#### Version
ArduPlane-release

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
fixed-wing

#### Hardware type
Pixhawk

Kiwa21, 2017-12-30:
That seems pretty interesting !
I was just about to start to do something similar for Copter.
Are you using a A* algorithms or your own ? Onboard ? What kind of costmap // occupancy map are you using ?

magicrub, 2017-12-30:
Hi @oneTimePad , thanks for the cool idea!

 Current infrastructure supports this behavior, sort of,, but it requires a small amount of work on the system generating the dynamic waypoints. Just write more waypoints to the end of the existing mission, jump to it, make sure the last command is a DO_JUMP back to where you came from.

Assumptions companion computer must know:
- current mission and/or length
- current waypoint index (to populate the DO_JUMP return index)

Current Cons:
- doing this a few times makes your mission grow and grow and grow so it puts a burden on the companion computer to erase the previous temporary mission. (this can happen after the DO_JUMP while executing the normal mission)

New method Pros:
- current index gets saved automatically
- the equivalent of the DO_JUMP is handled automatically with the correct return index
- deleting the temporary mission would happen automatically

Cons no matter what:
- The GCS is not notified of this mission change (the GCS needs to be notified, not too hard)
- what if during the temporary mission you switch to FBWA then back to AUTO?

I'd say this is a good devcall topic. I welcome you to join the call and discuss it, we could probably brainstorm a really awesome solution here! Who knows, maybe the right solution is this new one you proposed!

Although I'd suggest renaming MAV_CMD_INJECT_WAYPOINT to something like MAV_CMD_NAV_WAYPOINT_APPEND or maybe MAV_CMD_NAV_WAYPOINT_TEMP

oneTimePad, 2017-12-30:
@Kiwa21  Thank you!

For now the path-planning will be done offboard from the GCS. We are looking into utilizing Sample-based planning (RRT,RRT*, Greedy-biased RRT, and it's variants). We hope to put it onboard eventually. However we would still use a separate computer running the path-planning algorithm and it would just send these MAVLink messages to ArduPillot. 

It would be really cool if someone could get Sample-based planning or even A* running on Ardupilot. 

Most of my modifications were in libraries (AP_Mission and GCS_MAVLink), with very few changes to the actual ArduPlane code.

In regards to costmaps, I would need to check with my team, but I don't think we will be using them. This is because our requirement is to avoid fictitious obstacles which are floating geometric shapes. There will be no real world obstacle in the way. We just simply need to avoid hitting these geometric shapes (both moving and stationary).

oneTimePad, 2017-12-30:
@magicrub  thank you! yes i would love to join. When will this be going on?

magicrub, 2017-12-30:
@oneTimePad 
http://ardupilot.org/dev/docs/ardupilot-mumble-server.html

Every Monday at 3pm (California time) but the next one is New years Day so lots of people are traveling so the audience will be much lower than normal. Various people are just hanging out on the mumble chat all the time. If you see @tridge or @peterbarker they would probably have something to say about your idea.

oneTimePad, 2017-12-30:
@magicrub  awesome! thank you!

magicrub, 2018-01-09:
Work-in-progress of the concept can be found here:
https://github.com/RUAIAA/ardupilot/commit/bc61c0584afbf50effbd28ff4c42fc9514ddb488

oneTimePad, 2018-01-13:
I cleaned up the commit history for the obstacle-avoidance branch posted above. Here is also a short video of waypoint injection in SITL: https://youtu.be/jZ8iWfkBh-g

magicrub, 2018-01-13:
Thanks for the video! Can you do us a favor and post what the original mission was, and then what mission your app sent, and then a brief trace of what the behavior was.. like what way point it jumped to afterward. Thx!

oneTimePad, 2018-01-14:
No problem! Yup, will do!

oneTimePad, 2018-01-16:
Hope this is good! 

I used the mission in Tools/autotest/ArduPlane-Missions/CMAC-toff-loop.txt . Injected waypoints start from a sequence number greater than the last waypoint in a mission.

[image]

magicrub, 2018-01-16:
.. but what did you inject? What did those look like?

oneTimePad, 2018-01-17:
@magicrub  updated my comment. Sorry about that. Hope this is better!

dominozor, 2019-12-25:
Hi, is this feature going to be merged to master any time soon?

amilcarlucas, 2021-09-22:
ping?