ArduPilot/ardupilot #7470: Reset Attitude Controller after EKF change

lthall, 2017-12-31:
#### Issue details
Change in EKF results in a shift in our estimate of the Inertial frame estimate. This basically means that the attitude controller sees a sudden change in it's attitude. This results in a step change that ripples through the controller and can cause anything from a twitch, sudden change in attitude or a momentary loss of control.

#### Version
Master

#### Platform
Copter

#### Airframe type
All

#### Hardware type
All

#### Logs
_Please provide a link to any relevant logs that show the issue_

https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/ekf_check.cpp#L196

We should probably make a generic function that takes a quaternion rotation representing the change in Inertial frame from before to after the reset.

Maxce, 2019-08-23:
I apologize if I write in the wrong place, first time I appeal to the ardupilot developers team, but I think that I just faced the problem of quad instability due to attitude reset.
One difference, this happened not on EKF swithing, but on Yaw Reset.

I described it in detail here:
[https://discuss.ardupilot.org/t/sudden-twitch-on-ekf-yaw-reset/46104](url)
i could attach log's if it's neded.

Looking through the code, I found out (as i understood) that reaction on EKF change and Yaw reset almost the same - ` attitude_control->inertial_frame_reset();`
but for Yaw reset it was changed by one of commit from
`attitude_control->shift_ef_yaw_target(ToDeg(yaw_angle_change_rad) * 100.0f);`

I'm just wondering why processing only Yaw axis has been replaced by reseting the entire inertial_frame as for EKF change.
May be I'm completely wrong in, could someone explain to me whether it is safer to process only the Yaw axis in case of strong instability of the compass.

OXINARF, 2019-08-23:
@Maxce This issues list is for feature requests and confirmed bugs. For user support you need to handle it in the forum (http://discuss.ardupilot.org), post the logs there in your post.

amilcarlucas, 2019-08-26:
Hasn't this reset issue been fixed by @tridge recently ?

OXINARF, 2019-08-26:
@amilcarlucas Can you point out where? I'm not aware of @tridge touching this area at all, but I might have missed it.

amilcarlucas, 2019-08-26:
I am refering to the changes that allow the EKF instance to change during flight without causing attitude and position jumps. https://github.com/ArduPilot/ardupilot/pull/11913 https://github.com/ArduPilot/ardupilot/pull/11731 https://github.com/ArduPilot/ardupilot/pull/11720 https://github.com/ArduPilot/ardupilot/pull/11705

OXINARF, 2019-08-26:
I don't believe any of that does what is wanted here.

closed 2021-08-16 by lthall