ArduPilot/ardupilot #7478: waf build fail

patrickelectric, 2018-01-02:
#### Issue details
```bash
./waf configure --board px4-v2
./waf sub
```

using arm-none-eabi-gcc 7.2.0-1

#### Version
(master) b5bdaa18ac1edbc884ef1c43cd168eb4c90298e1

#### Platform

- [ ] All
- [ ] AntennaTracker
- [ ] Copter
- [ ] Plane
- [ ] Rover
- [X] Submarine

#### Airframe type
vectored

#### Hardware type
px4-v2

#### Logs
```bash
 ./waf sub   
Waf: Entering directory `/mnt/hd/git/blue/ardupilot/build/px4-v2'
[ 5/15] Processing modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
[ 6/15] Processing uavcangen: modules/uavcan/dsdl/uavcan/CoarseOrientation.uavcan modules/uavcan/dsdl/uavcan/Timestamp.uavcan modules/uavcan/dsdl/uavcan/equipment/actuator/1010.ArrayCommand.uavcan modules/uavcan/dsdl/uavcan/equipment/actuator/1011.Status.uavcan modules/uavcan/dsdl/uavcan/equipment/actuator/Command.uavcan modules/uavcan/dsdl/uavcan/equipment/ahrs/1000.Solution.uavcan modules/uavcan/dsdl/uavcan/equipment/ahrs/1001.MagneticFieldStrength.uavcan modules/uavcan/dsdl/uavcan/equipment/ahrs/1002.MagneticFieldStrength2.uavcan modules/uavcan/dsdl/uavcan/equipment/ahrs/1003.RawIMU.uavcan modules/uavcan/dsdl/uavcan/equipment/air_data/1020.TrueAirspeed.uavcan modules/uavcan/dsdl/uavcan/equipment/air_data/1021.IndicatedAirspeed.uavcan modules/uavcan/dsdl/uavcan/equipment/air_data/1025.AngleOfAttack.uavcan modules/uavcan/dsdl/uavcan/equipment/air_data/1026.Sideslip.uavcan modules/uavcan/dsdl/uavcan/equipment/air_data/1027.RawAirData.uavcan modules/uavcan/dsdl/uavcan/equipment/air_data/1028.StaticPressure.uavcan modules/uavcan/dsdl/uavcan/equipment/air_data/1029.StaticTemperature.uavcan modules/uavcan/dsdl/uavcan/equipment/camera_gimbal/1040.AngularCommand.uavcan modules/uavcan/dsdl/uavcan/equipment/camera_gimbal/1041.GEOPOICommand.uavcan modules/uavcan/dsdl/uavcan/equipment/camera_gimbal/1044.Status.uavcan modules/uavcan/dsdl/uavcan/equipment/camera_gimbal/Mode.uavcan modules/uavcan/dsdl/uavcan/equipment/device/1110.Temperature.uavcan modules/uavcan/dsdl/uavcan/equipment/esc/1030.RawCommand.uavcan modules/uavcan/dsdl/uavcan/equipment/esc/1031.RPMCommand.uavcan modules/uavcan/dsdl/uavcan/equipment/esc/1034.Status.uavcan modules/uavcan/dsdl/uavcan/equipment/gnss/1060.Fix.uavcan modules/uavcan/dsdl/uavcan/equipment/gnss/1061.Auxiliary.uavcan modules/uavcan/dsdl/uavcan/equipment/gnss/1062.RTCMStream.uavcan modules/uavcan/dsdl/uavcan/equipment/gnss/1063.Fix2.uavcan modules/uavcan/dsdl/uavcan/equipment/gnss/ECEFPositionVelocity.uavcan modules/uavcan/dsdl/uavcan/equipment/hardpoint/1070.Command.uavcan modules/uavcan/dsdl/uavcan/equipment/hardpoint/1071.Status.uavcan modules/uavcan/dsdl/uavcan/equipment/ice/1129.FuelTankStatus.uavcan modules/uavcan/dsdl/uavcan/equipment/ice/reciprocating/1120.Status.uavcan modules/uavcan/dsdl/uavcan/equipment/ice/reciprocating/CylinderStatus.uavcan modules/uavcan/dsdl/uavcan/equipment/indication/1080.BeepCommand.uavcan modules/uavcan/dsdl/uavcan/equipment/indication/1081.LightsCommand.uavcan modules/uavcan/dsdl/uavcan/equipment/indication/RGB565.uavcan modules/uavcan/dsdl/uavcan/equipment/indication/SingleLightCommand.uavcan modules/uavcan/dsdl/uavcan/equipment/power/1090.PrimaryPowerSupplyStatus.uavcan modules/uavcan/dsdl/uavcan/equipment/power/1091.CircuitStatus.uavcan modules/uavcan/dsdl/uavcan/equipment/power/1092.BatteryInfo.uavcan modules/uavcan/dsdl/uavcan/equipment/range_sensor/1050.Measurement.uavcan modules/uavcan/dsdl/uavcan/equipment/safety/1100.ArmingStatus.uavcan modules/uavcan/dsdl/uavcan/navigation/2000.GlobalNavigationSolution.uavcan modules/uavcan/dsdl/uavcan/protocol/1.GetNodeInfo.uavcan modules/uavcan/dsdl/uavcan/protocol/2.GetDataTypeInfo.uavcan modules/uavcan/dsdl/uavcan/protocol/341.NodeStatus.uavcan modules/uavcan/dsdl/uavcan/protocol/4.GetTransportStats.uavcan modules/uavcan/dsdl/uavcan/protocol/4.GlobalTimeSync.uavcan modules/uavcan/dsdl/uavcan/protocol/5.Panic.uavcan modules/uavcan/dsdl/uavcan/protocol/5.RestartNode.uavcan modules/uavcan/dsdl/uavcan/protocol/6.AccessCommandShell.uavcan modules/uavcan/dsdl/uavcan/protocol/CANIfaceStats.uavcan modules/uavcan/dsdl/uavcan/protocol/DataTypeKind.uavcan modules/uavcan/dsdl/uavcan/protocol/HardwareVersion.uavcan modules/uavcan/dsdl/uavcan/protocol/SoftwareVersion.uavcan modules/uavcan/dsdl/uavcan/protocol/debug/16370.KeyValue.uavcan modules/uavcan/dsdl/uavcan/protocol/debug/16383.LogMessage.uavcan modules/uavcan/dsdl/uavcan/protocol/debug/LogLevel.uavcan modules/uavcan/dsdl/uavcan/protocol/dynamic_node_id/1.Allocation.uavcan modules/uavcan/dsdl/uavcan/protocol/dynamic_node_id/server/30.AppendEntries.uavcan modules/uavcan/dsdl/uavcan/protocol/dynamic_node_id/server/31.RequestVote.uavcan modules/uavcan/dsdl/uavcan/protocol/dynamic_node_id/server/390.Discovery.uavcan modules/uavcan/dsdl/uavcan/protocol/dynamic_node_id/server/Entry.uavcan modules/uavcan/dsdl/uavcan/protocol/enumeration/15.Begin.uavcan modules/uavcan/dsdl/uavcan/protocol/enumeration/380.Indication.uavcan modules/uavcan/dsdl/uavcan/protocol/file/40.BeginFirmwareUpdate.uavcan modules/uavcan/dsdl/uavcan/protocol/file/45.GetInfo.uavcan modules/uavcan/dsdl/uavcan/protocol/file/46.GetDirectoryEntryInfo.uavcan modules/uavcan/dsdl/uavcan/protocol/file/47.Delete.uavcan modules/uavcan/dsdl/uavcan/protocol/file/48.Read.uavcan modules/uavcan/dsdl/uavcan/protocol/file/49.Write.uavcan modules/uavcan/dsdl/uavcan/protocol/file/EntryType.uavcan modules/uavcan/dsdl/uavcan/protocol/file/Error.uavcan modules/uavcan/dsdl/uavcan/protocol/file/Path.uavcan modules/uavcan/dsdl/uavcan/protocol/param/10.ExecuteOpcode.uavcan modules/uavcan/dsdl/uavcan/protocol/param/11.GetSet.uavcan modules/uavcan/dsdl/uavcan/protocol/param/Empty.uavcan modules/uavcan/dsdl/uavcan/protocol/param/NumericValue.uavcan modules/uavcan/dsdl/uavcan/protocol/param/Value.uavcan
[ 7/15] Creating build/px4-v2/ap_version.h
[ 8/15] CMake Configure px4
[ 9/15] PX4: Copying rc.APM to px4-extra-files/ROMFS/init.d/rc.APM
[10/15] PX4: Copying rc.error to px4-extra-files/ROMFS/init.d/rc.error
-- nuttx-px4fmu-v2-apm
[11/15] PX4: Copying rcS to px4-extra-files/ROMFS/init.d/rcS
[12/15] PX4: Copying startup to px4-extra-files/ROMFS/tones/startup
[13/15] PX4: Copying px4fmuv2_bl.bin to px4-extra-files/ROMFS/bootloader/fmu_bl.bin
-- could not find gdbtui
-- could not find ddd
-- could not find gdbtui
-- could not find ddd
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/bin/python (found version "3.6.4") 
-- GIT_DESC = fc46919934b82f3fa64a9b99f5ebe33de8bf6ab0
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/hd/git/blue/ardupilot/build/px4-v2/modules/PX4Firmware
INCLUDE_DIRS=/mnt/hd/git/blue/ardupilot/modules/PX4Firmware/src;/mnt/hd/git/blue/ardupilot/build/px4-v2/modules/PX4Firmware;/mnt/hd/git/blue/ardupilot/build/px4-v2/modules/PX4Firmware/src;/mnt/hd/git/blue/ardupilot/modules/PX4Firmware/src/modules;/mnt/hd/git/blue/ardupilot/modules/PX4Firmware/src/include;/mnt/hd/git/blue/ardupilot/modules/PX4Firmware/src/lib;/mnt/hd/git/blue/ardupilot/modules/PX4Firmware/src/platforms;/mnt/hd/git/blue/ardupilot/modules/PX4Firmware/src/drivers/boards/px4fmu-v2;/mnt/hd/git/blue/ardupilot/build/px4-v2/modules/PX4Firmware;/mnt/hd/git/blue/ardupilot/build/px4-v2/modules/PX4Firmware/src/modules/px4_messages;/mnt/hd/git/blue/ardupilot/build/px4-v2/modules/PX4Firmware/src/modules;/mnt/hd/git/blue/ardupilot/modules/PX4Firmware/mavlink/include/mavlink;/mnt/hd/git/blue/ardupilot/modules/PX4Firmware/src/lib/DriverFramework/framework/include;src/lib/matrix;/mnt/hd/git/blue/ardupilot/build/px4-v2/modules/PX4Firmware/px4fmu-v2/NuttX/nuttx-export/include;/mnt/hd/git/blue/ardupilot/build/px4-v2/modules/PX4Firmware/px4fmu-v2/NuttX/nuttx-export/include/cxx;/mnt/hd/git/blue/ardupilot/build/px4-v2/modules/PX4Firmware/px4fmu-v2/NuttX/nuttx-export/arch/chip;/mnt/hd/git/blue/ardupilot/build/px4-v2/modules/PX4Firmware/px4fmu-v2/NuttX/nuttx-export/arch/common
ADDING DRIVERS

[14/15] CMake Build px4 msg_gen
[2/4] Generating uORB topic headersValidating /mnt/hd/git/blue/ardupilot/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Parsing /mnt/hd/git/blue/ardupilot/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Note: message DATA64 is longer than 64 bytes long (74 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
[... cut ...] arch/chip', '-Imodules/PX4Firmware/px4fmu-v2/NuttX/nuttx-export/arch/common', '-DSKETCHBOOK="/mnt/hd/git/blue/ardupilot"', '-DCONFIG_HAL_BOARD=HAL_BOARD_PX4', '-DHAVE_OCLOEXEC=0', '-DHAVE_STD_NULLPTR_T=0', '-DUAVCAN_CPP_VERSION=UAVCAN_CPP03', '-DUAVCAN_NO_ASSERTIONS=1', '-DUAVCAN_NULLPTR=nullptr', '-DAPM_BUILD_DIRECTORY=APM_BUILD_ArduSub', '-DSKETCH="ArduSub"', '-DSKETCHNAME="ArduSub"', '../../ArduSub/AP_State.cpp', '-c', '-o/mnt/hd/git/blue/ardupilot/build/px4-v2/ArduSub/AP_State.cpp.16.o']
```

liang-tang, 2018-01-03:
rm build direction, and try again.

hsteinhaus, 2018-02-06:
Same problem here when building with gcc 7.2. Obviously, a 'const'  has been added to the built-in declaration of vnsscanf. After fixing this prototype in modules/PX4NuttX/nuttx/include/stdio.h and removing build dir, everything is fine again.

mhentschke, 2019-04-08:
Hi, I'm having a pretty similar issue on Ubuntu 19.04 with gcc 8.3 and can't seem to solve it. The declaration I have for vsscanf in my modules/PX4NuttX/nuttx/include/stdio.h is as follows: 
`int    vsscanf(char *buf, const char *s, va_list ap);`
Removing const from the *s declaration does not have any effect so far.
It seems, by the output that the first error waf gets is related to the absence of gdbtui but I can't seem to figure out why

khancyr, 2019-04-08:
hello,
we don't support Nuttx anymore... but use the official gcc 4.9 or gcc6 toolchain and it will work.
Closing this issue as Nuttx isn't support anymore

closed 2019-04-08 by khancyr

magicrub, 2019-04-08:
To repeat @khancyr  but with more detail:
NuttX (no longer supported):
`./waf configure --board px4-v2`

ChibiOS (this is better):
`./waf configure --board fmuv2`

but really you should probably be using `fmuv2 `by now (aka px4-v3)