ArduPilot/ardupilot #7481: ArduBoat: add support for vectored motor/thruster configuration

jaxxzer, 2018-01-04:
Over the past year I've received at least 10 inquiries about using ArduSub for a surface vessel specifically because of the vectored motor configuration that we support. There have been some successes, but most people have an issue that there is no support for RC receivers in ArduSub.

DavidIngraham, 2018-01-12:
Do you see this best implemented by adding support for vectored thrusters to rover (ArduBoat) or adding RC support to Sub?

labomat, 2018-04-19:
Count me in as number 11 ...

JimiJoplin, 2018-05-07:
Hi, i need to create a boat with ardusub. 
I would like to know where I can find information on similar projects.
I need to create a frame configuration like the bluerov1, but i need to use only the thrusters 1,2 and 6.
It is correct to put the following configuration on the AP_Motors6DOF.cpp:
|Motor #	|Roll Factor	|Pitch Factor	|Yaw Factor	|Throttle Factor	|Forward Factor	|Lateral Factor| 
|1	|0	 	|0		|-1.0	 	|0		|1.0		|0 |
|2 	|0		|0		|1.0		|0		|1.0		|0 |
**|3	|0		|0		|0		|0		|0		|0 |
|4 	|0 		|0		|0		|0 		|0 		|0 |
|5	|0		|0		|0 		|0		|0 		|0 |**
|6	|-0.25 		|0 		|0 		|0 		|0 		|1.0 |
simply put to 0 all the factors of the motors that i don't use?
Thanks

rmackay9, 2018-05-07:
We are getting closer to adding support for lateral thrust to Rover.  [This PR is the first new frame that allows lateral movement](https://github.com/ArduPilot/ardupilot/pull/8317).  There will be a follow PR in a few weeks that takes advantage of the lateral movement in all modes.

jaxxzer, 2018-05-08:
> Do you see this best implemented by adding support for vectored thrusters to rover (ArduBoat) or adding RC support to Sub?

I do recommend adding the capability to Boat. Sub is not meant to be constrained to 2D movement or to be operated with an RC receiver.

labomat, 2018-11-01:
Anything new on this issue?

rmackay9, 2019-01-04:
@labomat, as mentioned on the other thread, the static thruster support is in master and will go out with Rover-3.5 but multiple vectored motors are not supported - at least not in cases where the motors might be requested to point in different directions from each other.

To get new features added to AP, it's mostly about finding a qualified developer to make the changes.  That developer might be you or a volunteer in the community who is interested in helping or it can be a [paid developer](http://ardupilot.org/ardupilot/docs/common-commercial-support.html).