ArduPilot/ardupilot #7485: Understanding ekf2 of px4 firmware

ajaybhatt003, 2018-01-04:
Hello,
I am a graduate student. I have just started working on ekf2 (https://github.com/PX4/Firmware/blob/master/src/modules/ekf2/ekf2_main.cpp) code of px4 firmware. Kindly can you help me to understand it.
I am currently referring to this site http://ardupilot.org/dev/docs/ekf2-estimation-system.html but not sure that the same code is written in cpp as ekf2 for px4 firmware.
Please reply when you get time if possible provide me literature which explains about ekf2 code so that I can convert it to matlab code for my purpose.
Thanks in advance.

auturgy, 2018-01-04:
They’re not identical but have some commonality - the author is the same. For the derivation of the px4 ekf, look at Paul’s work here: https://github.com/priseborough/ecl/tree/master/matlab

Regards,

James

auturgy, 2018-01-04:
Closed. Not an ArduPilot issue.

closed 2018-01-04 by auturgy

ajaybhatt003, 2018-01-04:
Thank you for prompt reply
Just last question that matlab code on https://github.com/priseborough/ecl/blob/master/matlab/scripts/Inertial%20Nav%20EKF/GenerateNavFilterEquations.m is the same code as ekf2 written in cpp with px4 firmware.
Thanks

auturgy, 2018-01-04:
This is the ArduPilot repo, which is a different project, separate from px4.
Someone in px4 may have an idea, so better to ask there.