ArduPilot/ardupilot #7491: ArduCopter AP_Arming: Kill switch stops green flash

lthall, 2018-01-05:
Pulling the kill switch (CHX Option 32) or the arming button should not stop or enable the green led from flashing.

Both of these cases should work the same.

Green flash for ready to arm with GPS
Blue flash for ready to arm without GPS.

Then to arm both the arming button needs to be pushed and the kill switch needs to be pulled.

peterbarker, 2018-01-05:
On Fri, 5 Jan 2018, Leonard Hall wrote:

> Pulling the kill switch (CHX Option 32) or the arming button should not
> stop or enable the green led from flashing.

Pulling the arming switch should definitely stop the green LED from 
flashing - it should go solid green!  Did you mean the safety button 
there?

> Both of these cases should work the same.
> 
> Green flash for ready to arm with GPS
> Blue flash for ready to arm without GPS.

> Then to arm both the arming button needs to be pushed and the kill switch
> needs to be pulled.

I agree.  You should be able to see if your vehicle is armable based on 
the flashing lights before making the vehicle dangerous with the 
interlock/estop/safety switch.

However - isn't it the case that currently the safety switch *does* give 
that behaviour?  i.e. (assuming you don't have an interlock active) your 
vehicle will flash blue or green even if the safety switch light is also 
flashing?

I think the solution here is to still run the checks in pre-arm (so the 
GCS will get messages about interlock/estop), but NOT fail the pre-arm 
checks if you have one or the other active.  The arming checks will get 
the same checks but they'll fail.  For consistency, the safety switch 
check will get a copy in pre-arm (so you will be told on the GCS you won't 
be able to arm) but will also not fail the pre-arm checks so the LED will 
flash green/blue). @rmackay9?

lthall, 2018-01-05:
Hi Peter, Thanks for your time!!

Sorry, the point I was trying to make is the Motor Interlock and safety switch should have the same behaviour. Both should not stop the aircraft from signaling that it is ready to be armed. Both should only need to be checked when we actually try to arm. Currently the light will go to green flashing only when you pull the Motor Interlock switch. So the user has to repeatedly pull the Motor Interlock switch until they see the light go green and then arm.

Just to be clear about the option 32 / Motor Interlock. The way it operates is it must be pulled to the killed position before you can arm the aircraft. You then must let it go before any motors start spinning. You also must pull it to the killed position before you can disarm the aircraft. (just in case you were mixing it up with the Motor Emergency Stop option that I am not so familiar with, 31 I think).

So I assume that the safety switch does the right thing but the Motor Interlock switch does not. Sorry I had the Motor Interlock active when I was testing.

Thanks again for your time and help Peter!!

R-Lefebvre, 2018-01-05:
I agree.

The flashing green light should indicate the navigation system is ready.  It should not be affected by the motor interlock of E-stop switches in any way. 

I use the Motor Interlock, and I have it on a 2 position switch, not a momentary.  So it's in the "Off" position until I'm ready to go hot.  I should not have to repeatedly push it to the "On" position and back Off just to see if the system will be ready to arm.

peterbarker, 2018-01-05:
On Fri, 5 Jan 2018, Leonard Hall wrote:

> Sorry, the point I was trying to make is the Motor Interlock and safety
> switch should have the same behaviour. Both should not stop the aircraft
> from signaling that it is ready to be armed. Both should only need to be
> checked when we actually try to arm. Currently the light will go to green

Yep.  And I think we should add the same behaviour for estop.

> Just to be clear about the option 32 / Motor Interlock. The way it operates
> is it must be pulled to the killed position before you can arm the
> aircraft. You then must let it go before any motors [...]

Yes, I'm not greatly familiar with the interlock.  I've never used it, 
just seen it used.  I *do* use eStop on my quads.

> So I assume that the safety switch does the right thing but the Motor
> Interlock switch does not. Sorry I had the Motor Interlock active when I
> was testing.

So... I *think* that this patch would actually fix the problem you've got: 
https://github.com/ArduPilot/ardupilot/pull/7495

I'd like to do a huge amount more work in AP_Arming, actually.  Any 
extensive work I do to fix eStop behaviour / safety-switch behaviour is 
going to conflict horribly with PRs I've already got in place, however, so 
I think I'll leave that for later...

> Thanks again for your time and help Peter!!

No worries :-)

peterbarker, 2018-01-05:
On Fri, 5 Jan 2018, Robert Lefebvre wrote:

> The flashing green light should indicate the navigation system is ready. It
> should not be affected by the motor interlock of E-stop switches in any
> way.

Yes, but it should also reflect *every other* prearm check we do - not 
just whether you could navigate if required :-)  Fences would be one 
obvious example.

> I use the Motor Interlock, and I have it on a 2 position switch, not a
> momentary. So it's in the "Off" position until I'm ready to go hot. I
> should not have to repeatedly push it to the "On" position and back Off
> just to see if the system will be ready to arm.

Yes, that behaviour is silly IMHO.

lthall, 2018-01-06:
Thanks Peter!

rmackay9, 2018-01-08:
This is fixed in master now with this commit: https://github.com/ArduPilot/ardupilot/commit/077b0627012f2d998fae61a201f0dbf7c13cf823

closed 2018-01-08 by rmackay9