ArduPilot/ardupilot #7494: Arduplane flare over Land WP on incorrectly programmed mission (much higher altitude)

suraqis, 2018-01-05:
#### Issue details
If landing sequence programmed in incorrect way - last WP before Land WP is much higher than possible glide slope allows, than Arduplane will flare anyway over Land WP. I.e. if the last WP before land is on 1000 m AGL and in 100 m distance from Land WP, than UAV can't do the glideslope down to Land point and will fly over Land WP at high altitude, but flare will be engaged anyway. Maybe this situation should be managed in some way? Maybe just engaging RTL if altitude over Land WP is higher than say 30 m?

#### Version
3.8.2

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Fixed wing

#### Hardware type
SITL

#### Logs
Easily reproducible

magicrub, 2018-01-05:
Hmm, I think the issue here is why is it not doing a proper landing? Was it a (massive) alt drift or was it an incorrectly planned mission?

In order for it to safely RTL at >= 30m instead of flaring, it would have to know it's up at 31m. If it knows that, then it would already be trying to acheive the land approach.

There's already [LAND_ABORT_DEG](http://ardupilot.org/plane/docs/parameters.html#land-abort-deg-landing-auto-abort-slope-threshold) which requires a rangefinder to determine the approach angle is unacheivable and do an aborted landing go-around. But it sounds like a go-around would not help because you'd just repeat the same 31m above landing scenario again.

Perhaps a better solution is to sanity check the mission (issue https://github.com/ArduPilot/ardupilot/issues/4701) as a pre-arm?

suraqis, 2018-01-05:
In my case It was intentionally incorrect mission. After some autoland testings I was wondering what will happen if mission will be bad planned and glide slope to Land point will be impossible. I start SITL and made incorrect mission. As far as I know SITL has no rangefinder. So after passing over Land point Arduplane flared at high altitude and land few hundred meters behind Land point.

Mission sanity check is a good idea, but this check may be done only with flight model aware approach, I think. Also it shouldn't be pre-arm check, but rather pre-auto check or even after-upload check, because we can upload mission while in air.

geofrancis, 2018-02-04:
I have done this before when planning a mission, it was all at 100m but i forgot to set the landing alt to zero. 

The result was it flared at 100m, went into a flat spin and totalled the plane.

So some kind of sanity check on the landing altitude would be a good idea.