ArduPilot/ardupilot #7496: ArduPlane: handling servo failures

expntly, 2018-01-06:
#### Issue details
I posted on [discuss](https://discuss.ardupilot.org/t/how-to-tell-a-servo-goes-bad-in-flight/24334) but didn't hear anything.

Controls are critical for a plane.  Does ArduPilot have support for handling servo failures?  Is this a UAVCAN thing?  It would be great to at least know on the ground/GCS that a servo failed in flight and be able to take corrective actions.

#### Version
ArduPlane 3.8

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Plane

#### Hardware type
PH2.1

geofrancis, 2018-01-07:
I think the main problem is that there are so many different failure modes and different airframes would need to take different corrective action, to get redundancy on a standard airframe you really just need an extra elevator servo since most planes already have 2 aileron servos and you can fly without a rudder. 

for wings you need a pair of servos on each wing, by having a split elevon you should retain limited control with a servo failure.

magicrub, 2018-01-07:
For minor failure, the PIDs will compensate. For major failures your aircraft is going to have a hard time flying...

peterbarker, 2018-01-08:
On Fri, 5 Jan 2018, expntly wrote:

> Controls are critical for a plane. Does ArduPilot have support for handling
> servo failures? Is this a UAVCAN thing? It would be great to at least know
> on the ground/GCS that a servo failed in flight and be able to take
> corrective actions.

Look for Ryal Beale's thesis.  It's also around as a 
plane-adaptive-control branch.

magicrub, 2018-01-08:
@peterbarker Here's Ryan Beall's Adaptive controller:
Thesis:
https://github.com/rbeall/nps-thesis/blob/master/thesis.pdf

Presentation:
https://www.youtube.com/watch?v=_-baluMXZ-c

Source:
https://github.com/rbeall/ardupilot/tree/plane-adaptive-control3/libraries/ADAP_Control

lthall, 2018-03-18:
This is not a support forum.

closed 2018-03-18 by lthall