ArduPilot/ardupilot #7497: How to store previous parameter and then get it later?

hujin9190, 2018-01-06:
#### Issue details
I want to store a PWM value in a parameter and then get the value later. I followed the tutorial and created a parameter. Then, I used set_value and get_value functions to operate the value. However, it still gets zero each time rather than the set value. Can anyone help?

#### Version
It doesn't matter

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
It doesn't matter

#### Hardware type
pixhawk

hujin9190, 2018-01-06:
I need to do some calculation during the flight and get the new PWM based on the previous one. Thus, I need to store the previous status value first. If anyone can help, I very appreciate it.

peterbarker, 2018-01-06:
On Sat, 6 Jan 2018, hujin9190 wrote:

> I need to do some calculation during the flight and get the new PWM based
> on the previous one. Thus, I need to store the previous status value first.
> If anyone can help, I very appreciate it.

Have a look at the way hover throttle is done.

WickedShell, 2018-06-25:
This is a support topic rather then a feature request/issue. Please take it to [gitter](gitter.im/ardupilot/ardupilot) or [discuss](https://discuss.ardupilot.org) if you still need more help.

closed 2018-06-25 by WickedShell