ArduPilot/ardupilot #7499: ArduPlane: aggressive throttle changes when straight and level

expntly, 2018-01-07:
#### Issue details
During flight in AUTO cruise and on a straight and level leg, I noticed our current goes up and down by quite a bit all the time.  Looking at RCOU.C3 shows me the FC keeps changing the throttle, rather aggressively.  Very little wind.

Here's an example showing 20A+ jumps within 1-2s:

[image]

This seems inefficient.

Is it trying too hard to keep the airspeed constant?

For what it's worth I use

> THR_SLEWRATE=20

#### Version
ArduPlane 3.8.2

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Fixed wing

#### Hardware type
PH2.1

magicrub, 2018-01-07:
With THR_SLEWRATE=20 then your throttle is not capable of changing very fast. Heavily slewed. The current chasing stuff is often a sign of other tuning issues needing tweaking, such as the pitch. Have you played with the TECS controller much?

magicrub, 2018-01-08:
Without a log, I'm guessing.. but generally speaking if you do then it's using the throttle to maintain airspeed. If airspeed is sloppy then throttle will be sloppy (and vice versa). There's a coupling there. Could be sensor problem or it could be pitch/throttle/tuning settings.

expntly, 2018-01-11:
@magicrub 

Here's a log (250MB): https://drive.google.com/open?id=1ls9yVhjSz35V6mwTIUgS5fO6wtWALJdg

Haven't played much with TECS, but that might be a good idea.  I just stumbled upon TECS_THR_DAMP, currently at default 0.5.  I'll try and see if that helps in the meantime.  Thanks!

expntly, 2018-01-16:
@magicrub 

So I've been able to reproduce with SITL.

I've played with a few TECS parameters and TECS_TIME_CONST seems to help but I feel like it's hiding the underlying issue.

Here's with TECS_TIME_CONST=5 (default): 
[image]
Log: [00000008.BIN.zip](https://github.com/ArduPilot/ardupilot/files/1633617/00000008.BIN.zip)

Here's  with TECS_TIME_CONST=10: 
[image]
Log: [00000009.BIN.zip](https://github.com/ArduPilot/ardupilot/files/1633618/00000009.BIN.zip)

Plotting pitch related data seems to be quite correlated, but I can't figure out which direction the causality goes.

Any ideas?

magicrub, 2018-01-16:
@expntly Well done! Having a repeatable SITL test is surely the way to help us track this down. Thank you!

expntly, 2018-01-19:
I'd be curious to see if other people have seen it in their logs, especially on a "long" straight and level flight?

SlipstreamFPV, 2018-03-12:
I've had this issue too. It's very hard to get AUTO and CRUISE to fly efficiently as the motor is constantly accelerating. Good work expntly, hopefully the devs can track it down now.

magicrub, 2018-03-13:
I'm terribly sorry we haven't looked into this yet. Thank you @expntly for the logs. We'll get to it!

expntly, 2018-03-17:
We're flying with TECS_TIME_CONST at 7 but we're still seeing the throttle all over the place.

@magicrub Anything you'd like us to test?

expntly, 2018-05-20:
@magicrub Do you know if there's been commits around this issue?  Happy to go out there and try if there is.

Naterater, 2019-01-05:
This is my opinion that I have gained with testing data. Low-pitch props are VERY sensitive to airspeed and often make the current very noisy. Higher pitch props really level-out that current curve and prevent spikes. 

I can't say I have the same experience with the throttle changing so rapidly. Maybe we all need to just increase TECS_THR_DAMP. Mine is at 0.8, and my TECS time constant is 5.

30 minutes of data:
[image]

For 2 minutes of data:
[image]

I really think this may be a setup/support/tuning issue that doesn't require a code change. Maybe the defaults should change?

SlipstreamFPV, 2019-01-05:
Agreed.

IamPete1, 2021-08-30:
Since there has been no further reports and @Naterater thinks this we should probably not a code issue, I think we should close.

SlipstreamFPV, 2021-08-30:
Just to check, this isn’t related to the TECS issue with throttle input noise being amplified by the system leading to an oscillation of the throttle/pitch control loop?

IamPete1, 2021-09-11:
@SlipstreamFPV That sounds like a separate issue. Maybe you could open a new issue to track it if there is not one already.

closed 2021-09-11 by IamPete1