ArduPilot/ardupilot #7503: One motor may stop intermittently during flight when I set MOT_SPIN_MIN = MOT_SPIN_ARM = 0

gostil, 2018-01-08:
#### Issue details
I set MOT_SPIN_MIN = MOT_SPIN_ARM = 0 to start with no running motors. When I increase the throttle sometimes any one of the quad motors would not run, or stop running after a while. This is more apparent when I slightly increase the throttle, after which the rpm of the motors automatically starts increasing in the stabilize mode. 
#### Version
APM Copter V 3.5.4 Quad. (Frame type H)

#### Platform
[  ] Copter

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Quad H type

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
pixhawk,
#### Logs
_Please provide a link to any relevant logs that show the issue_

kd0aij, 2018-01-08:
Since MOT_SPIN_MIN is zero, shouldn't you expect this to happen?

lthall, 2018-01-08:
This is expected. I term build up will take one of the motors to zero again unless you are in the air.

closed 2018-01-08 by lthall