ArduPilot/ardupilot #7506: Compass: fill out _DEVID parameter descriptions with a compass name

rmackay9, 2018-01-09:
When users look at the COMPASS_DEV_ID (and DEV_ID2, DEV_ID3) all they see is a number.  We can make it much easier for users to know which compass the device id refers to by filling in the parameter descriptions with a long list of compass names using "@Values".

compass device id parameter location: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Compass/AP_Compass.cpp#L254
script to conert device-id to human readable name: https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/decode_devid.py

WickedShell, 2018-01-09:
That's a crazy long list to map out across the combinations (78 by my current count). Users need it so rarely that a little JS widget (IE convert the python script to JS) that maps the DEV_ID's out to human readable that we hosted would surely be better?

My objection comes from the fact that it's just an overwhelmingly long (and growing) list for a user to look up a value in.

OXINARF, 2018-01-09:
Yeah, I'm quite against this too, the list will be very long. The device id doesn't have only the device type, it has the bus type, the bus number and the device address too.

olliw42, 2018-01-12:
I too would have often wished that there is a bit more info available on which compass is which.
The long list might help, but might not actually be most user-friendly.
My feeling always was that, in e.g. MissionPlanner, the compass calibration page could offer that additional info. Specifically, for each compass it could also tell the DevID, and it's decomposition into what it is made off. As the latter could be done algorithmically (just reverse of how the DevId is constructed) it wouldn't need big lists, nor would it expose the user to big lists. At the end, this would be relatively minor code changes in MissionPlanner.
just my 2 cents

IamPete1, 2021-07-26:
The new MP compass screen decodes the ID's.

[image]

Maybe we should close this? @rmackay9 I'l leave it to you.

closed 2021-08-30 by IamPete1