ArduPilot/ardupilot #7507: Copter-3.5.3 SITL on MacOS: floating point error

aditya00j, 2018-01-09:
I have been working with Copter-3.5.3 release for the past couple of months. I was doing the development on Windows and everything was working ok. Today I shifted to MacOS and observed that I can't start SITL, it throws up a floating point error. This happens with the Copter-3.5.3 release. Below is the error text:

```
Started model + at -35.363261,149.165230,584,353 at speed 1.0
Starting sketch 'ArduCopter'
Starting SITL input
Using Irlock at port : 9005
bind port 5760 for 0
Serial port 0 on TCP port 5760
Waiting for connection ....
Smoothing reset at 0.001
Remaining: 0.250000 litres
Pump: 0.000000 l/s
Spinner: 0.000000 rev/s
position_demand=0.000000 jaw_gap=30.000000 load=0.000000
Loaded defaults from /Users/aditya/Documents/GA3_ArduPilot/Tools/autotest/default_params/copter.parm
bind port 5762 for 2
Serial port 2 on TCP port 5762
bind port 5763 for 3
Serial port 3 on TCP port 5763
Loaded defaults from /Users/aditya/Documents/GA3_ArduPilot/Tools/autotest/default_params/copter.parm
ERROR: Floating point exception - aborting
Abort trap: 6
```

I am doing this with a fresh clone of the code, with all defaults. It has probably something to do with difference in floating point operations across different OS, since I didn't get this error in Windows. Any help to resolve the issue will be highly appreciated! Please let me know if you want any more information.

peterbarker, 2018-01-10:
On Mon, 8 Jan 2018, aditya00j wrote:

> ERROR: Floating point exception - aborting
> Abort trap: 6
>
> I am doing this with a fresh clone of the code, with all defaults. It has
> probably something to do with difference in floating point [...]

Please start SITL with `--gdb --debug`; once the exception occurs, type 
`bt` into the gdb window to generate a backtrace and paste it in here. 
Ta!

aditya00j, 2018-01-10:
Apple moved from gdb to lldb... so I installed gdb and tried the debug option (`sim_vehicle.py --gdb --debug`), but I'm getting a bunch of gdb errors now :/

```
AJ:~ aditya$ gdb -x /var/folders/lw/g1_nkdgx5h3713rqnh8n6fc00000gn/T/tmpho6wqa 
--args /Users/aditya/Documents/GA3_ArduPilot/build/sitl-debug/bin/arducopter -S
 -I0 --home -35.363261,149.165230,584,353 --model + --speedup 1 --defaults /Use
rs/aditya/Documents/GA3_ArduPilot/Tools/autotest/default_params/copter.parm 
GNU gdb (GDB) 8.0.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-apple-darwin17.0.0".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /Users/aditya/Documents/GA3_ArduPilot/build/sitl-debug/bin/arducopter...
warning: can't find symbol '_ZN6Copter11home_is_setEv' in minsymtab
warning: can't find symbol '_ZN11AP_Airspeed12get_pressureEv' in minsymtab
...
```

Has anyone successfully used gdb with macOS High Sierra (v10.13)?

aditya00j, 2018-01-10:
This `warning: can't find symbol` happens and then I get the following message:

```
/var/folders/lw/g1_nkdgx5h3713rqnh8n6fc00000gn/T/tmpCUc_HN:1: Error in sourced command file:
During startup program terminated with signal ?, Unknown signal.
```

aditya00j, 2018-01-22:
I discovered that this is an issue with gdb and new macOS Sierra (I'm using v10.13.2)... has anyone successfully used gdb on Mac recently?

Effectively, SITL is not working on MacOS currently, at least with Copter. The SITL stops with an FPE after Gyro initialisation. I've tried changing default params, default starting location etc. Unfortunately I don't have a lot of experience in debugging, so any help would be appreciated!

peterbarker, 2018-01-22:
On Sun, 21 Jan 2018, aditya00j wrote:

> I discovered that this is an issue with gdb and new macOS Sierra (I'm using v10.13.2)... has anyone successfully used gdb on Mac recently?

What compiler are you using?

aditya00j, 2018-01-22:
I'm using the standard toolchain for MacOS (so, clang). Here is the complete response to `./waf configure --board sitl`:

```
Setting top to                           : /Users/aditya/Documents/GA3_ArduPilot 
Setting out to                           : /Users/aditya/Documents/GA3_ArduPilot/build 
Autoconfiguration                        : enabled 
Setting board to                         : sitl 
Checking for 'clang++' (C++ compiler)    : /usr/bin/clang++ 
Checking for 'clang' (C compiler)        : /usr/bin/clang 
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes 
Checking for header endian.h                   : yes 
Checking for header byteswap.h                 : not found 
Checking for program 'python'                  : /Users/aditya/anaconda/bin/python 
Checking for python version >= 2.7.0           : 2.7.13 
Checking for program 'python'                  : /Users/aditya/anaconda/bin/python 
Checking for python version >= 2.7.0           : 2.7.13 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/local/bin/git 
Checking for program 'size'                    : /usr/bin/size 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
'configure' finished successfully (1.307s)
```

auturgy, 2018-01-22:
Ref https://github.com/ArduPilot/ardupilot/pull/7578

aditya00j, 2018-01-22:
Great, using g++ did the trick!

Closing this issue, thanks for your help!

closed 2018-01-22 by aditya00j

peterbarker, 2018-01-23:
On Mon, 22 Jan 2018, aditya00j wrote:

> Closed #7507.

FWIW the same happens under Linux ATM :-)

bart, 2019-02-17:
Unfortunately this won't work for me. Still having this issue, even with g++ :( Any solution? I'm on OSX Mojave

lvale, 2019-02-19:
[image]

using these commands

```
 ./waf distclean
git submodule deinit -f . 
git clean -f -x -d 
git pull 
git submodule update --init --recursive 
./Tools/autotest/sim_vehicle.py -v ArduCopter -f heli -L TagusPark

```

bart, 2019-02-19:
@lvale Which version of gcc / g++ are you using on Mac and which branch did you checkout?

lvale, 2019-02-19:
for SITL, plain clang

>Moon:~ luis$ gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 10.0.0 (clang-1000.11.45.5)
Target: x86_64-apple-darwin18.2.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

on ArduPilot, working on master

bart, 2019-02-19:
All right, cool. I'll try it out!

peterbarker, 2019-02-20:
Out of interest, @lvale does it currently work on Copter's stable branch?

bart, 2019-02-20:
Also another question which is location related. I tried to start it with parameter `-L 3DRBerkeley` but it always starts in TagusPark, even after completely removing ardupilot directory and re-initilize everything. Is that a bug or must I reset it somehow?

peterbarker, 2019-02-20:
@bart Looks correct to me.

[image]

Started like this:
`pbarker@bluebottle:~/rc/ardupilot(master)$ ./Tools/autotest/sim_vehicle.py -L 3DRBerkeley -v ArduCopter`

bart, 2019-02-20:
Hey @peterbarker I did, but the vehicle always stays at TagusPark.

```
# sim_vehicle.py -v ArduCopter -f quad -L 3DRBerkeley
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at 37.872991,-122.302348,20,260 (3DRBerkeley)
SIM_VEHICLE: WAF build
[...]
SIM_VEHICLE: Run ArduCopter
SIM_VEHICLE: "/Users/bart/ardupilot/Tools/autotest/run_in_terminal_window.sh" "ArduCopter" "/Users/bart/ardupilot/build/sitl/bin/arducopter" "-S" "-I0" "--home" "37.872991,-122.302348,20,260" "--model" "+" "--speedup" "1" "--defaults" "/Users/bart/ardupilot/Tools/autotest/default_params/copter.parm"
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--out" "127.0.0.1:14550" "--out" "127.0.0.1:14551"
RiTW: Starting ArduCopter : /Users/bart/ardupilot/build/sitl/bin/arducopter -S -I0 --home 37.872991,-122.302348,20,260 --model + --speedup 1 --defaults /Users/bart/ardupilot/Tools/autotest/default_params/copter.parm
```

To check it I did `STABILIZE> status` with the following output:
```
[...]
323: SIMSTATE {roll : 0.0, pitch : 0.0, yaw : 0.0, xacc : 0.0, yacc : 0.0, zacc : -9.80665016174, xgyro : 0.0, ygyro : 0.0, zgyro : 0.0, lat : 387401950, lng : -93018430}
[...]
```

In QGroundControl it looks like this:
[image]

bart, 2019-02-20:
I restarted my Macbook. Now it looks good!

lvale, 2019-02-20:
LoL.... welcome to my test flying field :)

lvale, 2019-02-20:
> Out of interest, @lvale does it currently work on Copter's stable branch?

yes. Sent you a bunch of stuff that clang still complains :)

peterbarker, 2019-02-20:
On Wed, 20 Feb 2019, Rene Bartkowiak wrote:

> I restarted my Macbook. Now it looks good!

My guess is that there was a rogue ArduPilot process still running on the 
machine, and that the new instance was actually dying due to being unable 
to bind the port.  Next time this happens, try "ps wauxf | grep- i ardu" 
and see if there are ardupilot processes floating around...

peterbarker, 2019-02-20:
On Wed, 20 Feb 2019, Rene Bartkowiak wrote:

> To check it I did STABILIZE> status with the following output:
> 
> [...]
> 323: SIMSTATE {roll : 0.0, pitch : 0.0, yaw : 0.0, xacc : 0.0, yacc : 0.0, zacc : -9.80665016174, xgyro : 0.0, ygyro : 0.0, zgyro : 0.0, lat : 387401950, 
> lng : -93018430}
> [...]

status takes an argument - the message you want to look at.  e.g.:

status SIMSTATE

bart, 2019-02-21:
Hey @peterbarker thanks a lot for your answers. I did the `ps aux` check already before and there was no SITL task running. Whenever I start it the first time after reboot with a location that location is set for all upcoming sessions. Just after reboot I can set a new one but that's fine, because I just needed one special location.

ddomit, 2019-04-25:
Where can I get the Copter-3.5.3 SITL Binary? I tried downloading it from the dronekit source but there is only copter 3.3.

Thanks!

peterbarker, 2019-04-25:
On Wed, 24 Apr 2019, ddomit wrote:

> Where can I get the Copter-3.5.3 SITL Binary? I tried downloading it from the dronekit source but there is only copter 3.3.

We only produce Linux x86 and arm binaries for SITL - if you want those 
you can find them on firmware.ardupilot.org

Otherwise you'll need to compile SITL binaries yourself and set some 
environment variables so dronekit-sitl picks them up.

SITL_BINARY
SITL_DEFAULTS_FILEPATH