ArduPilot/ardupilot #7509: Plane: Option to use Q_TRANS_DECCEL for VTOL_LAND in mission landing plan

DavidIngraham, 2018-01-09:
#### Issue details
Quadplanes use the parameter Q_TRANS_DECCEL to calculate the required transition distance for a VTOL Landing when completing a Q_RTL. However, when a mission ends with a waypoint followed by a VTOL_LAND, the vehicle will immediately transition after the waypoint, and fly as a VTOL to the landing point.

Generally, this is an acceptable behavior because users can choose to terminate their mission either way, depending on their preference of the two options:

1. Explicitly choose the transition point and land with VTOL_LAND
2. Allow the vehicle to determine its own transition point by landing with a Q_RTL

However, with this current set of options, it is not possible to program a DO_LAND_START landing plan that uses Q_TRANS_DECCEL to determine the transition point for landing. The only way to land with Q_TRANS_DECEL is to do an RTL with Q_RTL_MODE=1. Ending a DO_LAND_START sequence with an RTL will create an infinite loop.

For some larger quadplanes, there is very little reserve power available for transition and landing, so it is important to transition as late as possible (accounting for any headwind). Additionally, some missions require that the vehicle approach the VTOL landing from a specific heading or perform a LOITER_TO_ALT before landing to lose altitude (either of these require an explicit landing plan).

Because of this, we would like to be able to specify in a parameter of the VTOL_LAND mission item that the vehicle should use the same logic as Q_RTL to determine when to transition back to VTOL.

#### Version
Plane 3.8.4+

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Any VTOL Plane 

#### Hardware type
N/A

#### Logs
N/A

frizensami, 2018-01-11:
I would really like this behavior as well :+1: 

This will let users with a properly configured Q_TRANS_DECEL avoid setting the exact transition point during missions with a VTOL_LAND.

tridge, 2018-11-11:
I've assigned to @WickedShell and myself
I'm actually tempted to make using Q_TRANS_DECEL the default for NAV_VTOL_LAND in a future version. I think it could be built on top of the recently merged #9707
What we really need is the feature from #9707, but without the "circle and come into the wind" part of it.
Michael, what do you think? We could rename the OPTION_MISSION_LAND_FW_APPROACH to OPTION_MISSION_LAND_INTO_WIND, and make that option only affect whether we come into the wind or not. If not set then the plane would fly directly towards the landing point in fixed wing mode, using TECS_LAND_ARSPD as target airspeed, and transition at the Q_TRANS_DECEL distance.

DavidIngraham, 2018-12-02:
That sounds very reasonable to me.

IamPete1, 2021-07-12:
This has now been implemented in https://github.com/ArduPilot/ardupilot/pull/14938

closed 2021-07-12 by IamPete1