ArduPilot/ardupilot #7510: variable is not used, confused.

ender-chen, 2018-01-09:
#### Issue details
_Please describe the problem, or desired feature_
Recently I read the AP_GPS.cpp, In calc_blended_state() function.
```
    blended_alt_offset_cm = 0.0f;
    blended_NE_offset_m.zero();
    for (uint8_t i=0; i<GPS_MAX_RECEIVERS; i++) {
        if (_blend_weights[i] > 0.0f) {
            blended_NE_offset_m += location_diff(state[GPS_BLENDED_INSTANCE].location, corrected_location[i]) * _blend_weights[i];
            blended_alt_offset_cm += (float)(corrected_location[i].alt - state[GPS_BLENDED_INSTANCE].location.alt) * _blend_weights[i];
        }
    }
```
 the variable blended_alt_offset_cm is set, but when we use it next time, it will not used.
```
Vector2f blended_NE_offset_m;
    float blended_alt_offset_cm = 0.0f;
    blended_NE_offset_m.zero();
    for (uint8_t i=0; i<GPS_MAX_RECEIVERS; i++) {
        if (_blend_weights[i] > 0.0f && i != best_index) {
            blended_NE_offset_m += location_diff(state[GPS_BLENDED_INSTANCE].location, state[i].location) * _blend_weights[i];
            blended_alt_offset_cm += (float)(state[i].location.alt - state[GPS_BLENDED_INSTANCE].location.alt) * _blend_weights[i];
        }
    }
```
I don't think it works well. 

#### Version
3.5.4

#### Platform

[  ] Copter

OXINARF, 2018-01-09:
@ender-chen If you use links to specific lines it's more easy to understand. I guess what you are saying is that after doing

https://github.com/ArduPilot/ardupilot/blob/c56529f39cdb38a33aabef939a7e460c5e1c9961/libraries/AP_GPS/AP_GPS.cpp#L1460-L1468

we don't use it anywhere. Is that correct?

magicrub, 2018-01-09:
Yup, looks like a copy/paste error that should be removed. Luckily it was totally harmless. @WickedShell should chime in on this just in case we're missing something

WickedShell, 2018-01-09:
Looks to be correct sent in a PR that closes it.

magicrub, 2018-01-09:
merged!

OXINARF, 2018-01-09:
I've re-opened this since I don't believe the merged PR is the correct fix.

ender-chen, 2018-01-10:
@OXINARF Yeah that is just what I mean.
But how can you get that link? I don't know how to do it.

OXINARF, 2018-01-10:
@ender-chen https://help.github.com/articles/creating-a-permanent-link-to-a-code-snippet/

ender-chen, 2018-01-10:
@OXINARF  Oh, I get it, thank you.

IamPete1, 2021-07-26:
Looks like this is fixed in master, please re-open if I'm wrong.

PR #7514 (2018-01-09): AP_GPS: Remove unused blending math
Thanks to @ender-chen for noticing, closes #7510
commit 9e37e9ee (2018-01-09): AP_GPS: Remove unused blending function.

Thanks to ender-chen for noticing, closes #7510
closed 2021-07-26 by IamPete1