ArduPilot/ardupilot #7515: UDP telemetry - wildcard chars in IP

lospalos, 2018-01-09:
#### Issue details
With AP builds for linux there is possibility to connect to FC via UDP. The parameter for process is 
- A udp:IP-address:port 
IP-address is the one from which is alloved connection to FC process. There is no possibility to set the address with wildcard number, e.g. 192.168.1.0 or 192.168.1.* and it is bit annoying to have there only fixed address. If the FC and monitoring computer are on the same local net (xxx.xxx.xxx.0) they could get IP address dynamicaly from DHCP. If the address of monitoring computer is always changed, user have to always connect to FC via terminal and edit parameter -A in starting script .
With wildcard IP-address possibility this would be much easier.

#### Version
_What version was the issue encountered with_

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
Raspberry Pi, all HW running linux

#### Logs

muramura, 2018-01-10:
I think that name resolution function is necessary. For example rarp protocol.

khancyr, 2018-01-10:
That is how UDP works, you need to set where you want to send data ! Maybe a broadcast logic and fix to first client can be implemented.
Otherwise, you should get you ip and client ip before launching ardupilot instance in the startup script you are using

OXINARF, 2018-01-10:
We already have the broadcast logic, use `udp:192.168.1.255:14550:bcast` and let us know if it works for you.

lospalos, 2018-01-10:
@OXINARF
Great, it really works :-) and it makes my life muuuuch easier.
I didn't know it since I 'can't see it anywhere in documentation (mean docu from Erle, Even from Emlid)

OXINARF, 2018-01-11:
Great, it means we can close this!

Sometimes documentation lacks behind as we don't have enough man-power to do it - also, this isn't something most users mess up with.

closed 2018-01-11 by OXINARF