ArduPilot/ardupilot #7516: Copter: Totally incorrect mode selected when poor RC reception

SergeyBokhantsev, 2018-01-09:
#### Issue details
This is a very dangerous issue happened to me. Flying on the border of RC range I engaged RTL.
Then as copter returned to RC zone RC reception restored. But there was a delay in RCIN Ch.5 for some reason. It was still low, so Stabilize mode engaged unexpectedly for a few seconds.

[image]

#### Version
3.5.4

#### Platform
[ x ] Copter

#### Airframe type
X4

#### Hardware type
PixRacer + FrSky X8R receiver (configured as "no pulses on RC lost")

Pedals2Paddles, 2018-01-09:
That would indicate the delay was in your RC receiver, not ArduCopter.  It would probably be best to configure your receiver to have pitch, roll, and yaw channels go to neutral, CH5 stay unchanged, and CH3 go below the failsafe level.

SergeyBokhantsev, 2018-01-09:
@Pedals2Paddles my RX is configured to no-pulses at signal lost. I thought it is the modern way to handle failsafe. 
The problem is when the SBUS feed is restored, Ch 5 has a delay. It is not always a delay exists on RC signal restore but if it exists - then it is always only on Ch.5. This is strange. 
Could it be a faulty receiver?

Anyway, when Stabilize was engaged in my case above, the Ch5 value was only 874. This is way lower than calibrated range. May be this could be an indicator of the signal validity.

Pedals2Paddles, 2018-01-10:
You've opened three issues for the same thing now. You have an incorrectly configured RC receiver, or a malfunctioning RC receiver.  This is not an issue/bug in ArduCopter. The discuss forum would be a better place to ask for help with your RC receiver. https://discuss.ardupilot.org/

SergeyBokhantsev, 2018-01-10:
I don't know why you saying the tickets are same, they all about different things.
Here my point was that if the current level of the channel is lower that the minimal calibrated value then such value must be ignored.

Pedals2Paddles, 2018-01-10:
That is not correct.  Please see the wiki on CH5 flight mode selection.  Flight mode 1 is PWM 0-1200.  What you experienced is expected behavior due to an incorrectly configured or malfunctioning RC receiver.  Support on better ways to configure your RC receiver should be posted in the discuss forum since this is not a bug.

OXINARF, 2018-01-10:
I agree with @Pedals2Paddles, this is an issue with your receiver.

> Here my point was that if the current level of the channel is lower that the minimal calibrated value then such value must be ignored.

Channel 5, as well as all others after it, aren't usually calibrated. We have defined intervals of PWM values that determine what they do. We could certainly respect the min/max values configured in the parameters, but there are very few users that will have them set - I'll change the title to reflect this possible improvement.

closed 2021-08-08 by IamPete1