ArduPilot/ardupilot #7517: Fly modes management critical issue: RTL was not triggered

SergeyBokhantsev, 2018-01-09:
#### Issue details
I have an RTL action set to Ch.7 Opt in MissionPlanner/Extended tuning.
Also I have an RTL as my failsafe action for RC.

Steps:
1. Engage LOITER by Ch.5
2. Fly to the border of RC range in LOITER, engage RTL by switching Ch.7 high on RX => RTL engages, OK.
3. Copter completely losts RC signal => RTL is already on, no flymode change, OK
Note at that point TX switches are: Ch.5 - LOITER, Ch.7 - RTL.
4. RC signal appears. Both Ch5 and Ch7 reports correct levels 

Actual result: LOITER mode engaged
Expected result: RTL mode

#### Version
3.5.4

#### Platform
[ x ] Copter

#### Airframe type
X4

#### Hardware type
PixRacer + FrSky X8R receiver (configured as "no pulses on RC lost")

Pedals2Paddles, 2018-01-10:
You've opened three issues for the same thing now. You have an incorrectly configured RC receiver, or a malfunctioning RC receiver.  This is not an issue/bug in ArduCopter. The discuss forum would be a better place to ask for help with your RC receiver. https://discuss.ardupilot.org/

OXINARF, 2018-01-10:
This issues list is for feature requests and confirmed bugs. For user support please post in the forum (http://discuss.ardupilot.org).

If this is with the same receiver (and it appears to be), then your receiver is sending a changed CH5 option, which is why it will change mode. Post a log on the forum and it'll be obvious.

closed 2018-01-10 by OXINARF

SergeyBokhantsev, 2018-01-10:
I cannot repro that behavior anymore by my steps. Not an issue.