ArduPilot/ardupilot #7519: Copter descended (and hit roof) for no reason instead of continue flying to next waypoint

SergeyBokhantsev, 2018-01-10:
#### Issue details
Auto mission with 40 waypoints. Copter reached Wp26 and going to wp27. But because of the poor RC reception there were several mode switches to Stabilize-RTL-Auto-RTL-Auto:
[image]

Finally Auto mode was on but copter did not move to any waypoint. 
Instead it just descended for no reason until it hitted the roof.

[image]

#### Version
ArduCopter 3.5.3

#### Platform
[ x ] Copter

#### Airframe type
X4

#### Hardware type
Pixracer

#### Logs
https://www.dropbox.com/s/bk3s2hrqcyho7zw/00000120_auto_land.BIN?dl=0

Pedals2Paddles, 2018-01-10:
You've opened three issues for the same thing now. You have an incorrectly configured RC receiver, or a malfunctioning RC receiver.  This is not an issue/bug in ArduCopter. The discuss forum would be a better place to ask for help with your RC receiver. https://discuss.ardupilot.org/

SergeyBokhantsev, 2018-01-10:
There is a log and there is a screenshot where copter descends in AUTO, while no descend in mission at that point.
I don't understand how receiver (faulty or not) could force ArduCopter to do a things that are not exists in mission.
I don't ask for a help. I just want ArduPilot to be reliable project and noted that there is a unexpected behavior. It is up to dev team to get to the bottom of this case or to close the ticket.

lthall, 2018-01-10:
@Pedals2Paddles This does look like strange behaviour.

@SergeyBokhantsev This flight appears to be one of the most irresponsible flights I have seen in a long time. I hope you had permission to fly like this and a good reason to do so! If not I would strongly suggest that you find out more about your local regulations. I would also suggest finding a less populated area to test.

OXINARF, 2018-01-11:
> Version
>
> ArduCopter 3.5.3

@SergeyBokhantsev You are not flying 3.5.3, you are flying a custom version. I've looked at your changes and they look harmless but don't hide this kind of fact again.

Now, why does the log end without a disarm? What happened?
Right after you changed to Auto, a radio failsafe happened again and was never cleared until the end of the log.

I'm also not sure of the state of the vehicle as at some point in the flight both your IMU accelerometers  started clipping, with the second one having a big lot of them.

Why do you have CMD log disabled? This way it's impossible to know what command Auto was running. It's possible it had jumped to the last Land command, but now we can't know.

peterbarker, 2018-01-11:
On Wed, 10 Jan 2018, Francisco Ferreira wrote:

> command Auto was running. It's possible it had jumped to the last Land
> command, but now we can't know.

Any tlog available?

SergeyBokhantsev, 2018-01-11:
@OXINARF as copter begin descending radio connection was lost. So RC failsafe was on. After copter have landed it autodisarmed.

I'm also saw these clipping and I don't know the reason. Such clippings happened before in other flights, but I didn't noticed they causes any problems.

I didn't know about CMD log. Sadly it was disabled.

@peterbarker no Tlog

rmackay9, 2018-01-11:
Not having looked at the logs but I'd guess that the altitude type was set to something the pilot wasn't intending.  For example set to an absolute altitude instead of a relative or terrain altitude.

rmackay9, 2018-01-11:
I think I will close this issue because the issues list is just for enhancement requests and confirmed bugs.  Sorry, we have to be a little strict about this or the issues list just turns into a support forum.  This item should probably have been raised here: https://discuss.ardupilot.org/c/arducopter/copter-3-5

closed 2018-01-11 by rmackay9

tizianofiorenzani, 2018-11-20:
I think this issue has not been fixed, as I posted in my post [here](https://discuss.ardupilot.org/t/desired-altitude-goes-to-zero-then-all-the-way-up-when-stabilize-guided/35212/9). A rapid switch MODE->STAB-> MODE causes the Desired altitude to go all the way to zero (at maximum descend rate) and then climb back at the initial altitude (at maximum climb rate). MODE looks like is any mode with controlled altitude (for example RTL or GUIDED). It happened to me in flight as I had a glitch in the SBUS

I tested this in Copter 3.5.7 and 3.6.0 and I am fairly sure is still in master.

tizianofiorenzani, 2018-11-20:
This can easily be replicated in SITL with a transmitter connected to MP. Set the vehicle to RTL, then rapidly switch to STABILIZE then RTL again. The desired altitude Goes to zero in STAB, then in RTL starts from slightly under the current altitude and keeps decreasing at maximum descend rate, reaches zero, then starts climbing again. After that the vehicle starts moving to the RTL location.