ArduPilot/ardupilot #7524: Copter: Additional Hex propeller configuration

lthall, 2018-01-11:
Add hex propeller configuration that may be less susceptible to motor failure:

To quote @tuloski -
Basically changing the standard configuration of hexa with alternating propellers rotation PNPNPN, using a configuration PPNNPN can ensure that in 4 failures you keep the controllability, in 2 failures you don't. It is very interesting to implement probably. And btw in the two cases where you don't have full controllability, you can still loose yaw control, but land safely having the thrust, roll and pitch.

https://arxiv.org/pdf/1403.5986.pdf

This may require come creative coefficient selection to achieve the best results.

abhayjindal51, 2018-02-16:
Hi,@Ithall I want to work on this issue.

lthall, 2019-07-18:
Hi @abhayjindal51 I just saw your reply as you got my name wrong :)

PR #16563 (2021-02-10): AP_Scripting: add fault tolerant hex MotorsMatrix example
This adds a second example of adding custiom mixers with FRAME_CLASS = 15. This adds the fault tolerant hex from https://github.com/ArduPilot/ardupilot/pull/10852 
Fixes #7524
closed 2021-02-10 by rmackay9