ArduPilot/ardupilot #7529: Synthetic airspeed not reported in VFR_HUD Mavlink any longer

Hwurzburg, 2018-01-11:
Sometime after 3.4 plane, this reporting was stopped. It is a useful OSD feature. Is there any reason why it was okay previously to report it and not now ? It would be a simple change to return it,I believe with absolutely no flight impact since its just display information.

OXINARF, 2018-01-11:
I'm not a Plane expert, but it looks like it does try to get (and send) an airspeed estimate:

https://github.com/ArduPilot/ardupilot/blob/d3d46a7f2c4183578aead4751812203f5fe5e855/ArduPlane/GCS_Mavlink.cpp#L250-L255

Hwurzburg, 2018-01-11:
The issue, I believe, is that the routine used :ahrs.airspeed_estimate() returns FALSE and does not set the aspeed  variable if no sensor is enabled..something is wrong here....the code shown confirms that no speed is reported if no airspeed sensor is enabled...

OXINARF, 2018-01-12:
Code doesn't say that:

https://github.com/ArduPilot/ardupilot/blob/b47d575f65b71589e8893e7d4b19d82057f534be/libraries/AP_AHRS/AP_AHRS_DCM.cpp#L972-L1002

As long as you have GPS it should give you an airspeed estimate.

Hwurzburg, 2018-01-12:
ahrs.airspeed_estimate() is the routine called to report in VFR_HUD report, not airspeed_estimate()....there is a synthetic estimate made and used for flight, but not reported over MAVLINK, which is the issue...

OXINARF, 2018-01-12:
@Hwurzburg I'm sorry but that shows a lack of knowledge of C++. Plane uses the AP_AHRS_NavEKF class, which calls the DCM method I showed above:

https://github.com/ArduPilot/ardupilot/blob/b47d575f65b71589e8893e7d4b19d82057f534be/libraries/AP_AHRS/AP_AHRS_NavEKF.cpp#L467-L472

Hwurzburg, 2018-01-12:
True, I am not a C++ programmer, sorry, but it still does not change the fact that airspeed is being reported to mine, and several others in the forums, OSDs as being 0. Someone suggested that TECS_SYNAIRSPEED needs to be enabled, but from my non-expert reading of the code, it seems only to affect auto-throttle and is really targeted at Quadplane transistions.

OXINARF, 2018-01-12:
It also doesn't change the fact that this issues list is for confirmed bugs and feature requests, which this isn't either (it's not even confirmed to be an ArduPilot problem, it can be an OSD problem).

For user support you should post in the forum (http://discuss.ardupilot.org), but I'll let this here for a few more days to see if we get a Plane maintainer to confirm my comments.

Hwurzburg, 2018-01-12:
Thanks, I did post in the forum...no responses....I posted on the OSD forum, and several people had the same issue....just to confirm its not an OSD issue, I took the OSD from this 3.8 Pixhawk system and put it in my 3.4 APM system with no airspeed sensor, and it correctly displays the synthetic airspeed(OSD is setup to show ground speed and airspeed)....this would tend to confirm its not an OSD issue...

tridge, 2018-01-12:
Airspeed is reported in VFR_HUD with no airspeed sensor enabled. I fly a plane regularly with that setup.
My guess is you have ARSPD_TYPE set, thus enabling an airspeed sensor, but don't have one attached. So the zero value is coming from a floating analog pin.
If you still can't work it out and add a link here to the issue you posted on the discussion forums, making sure you post a tlog.

closed 2018-01-12 by tridge

Hwurzburg, 2018-01-12:
Thanks! You are right....type is indeed set....in Mission Planner the default is to set this value, the drop down list has no option for "none"....also this explains why previous versions do not have this issue, type appears to have been added recently, ie 3.8.x....I will confirm in flight in the next few days...
Hopefully this does not impact stall prevention and is just a reporting issue, ie synthetic airspeed is still being calculated from gps and used...but perhaps not..

I suggest this issue be turned into at least an enhancement request,ie Sensor type should default to NONE and that option be shown in the Mission planner's drop down box, to avoid this problem in the future.

I will report if that fixed it as soon as I can get out and fly again...thanks again

Hwurzburg, 2018-01-14:
Verified that this was the issue...I have requested a MP fix or enhancement