ArduPilot/ardupilot #7535: Dual Reverse Thrust

Naterater, 2018-01-14:
#### Issue details
Please add or confirm that reverse thrust is compatible with differential thrust. Twin engine planes are becoming very common, and I think that this could help yaw stability at low airspeeds when the rudder is less effective. This could prove to be very useful if more control is needed for precise landings as well.

#### Version
3.8.x

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Classic Twin

#### Hardware type
Pixhawk 2.1

#### Logs
N/A

iskess, 2018-01-14:
Just a quick PSA,
Be careful with reverse thrust on a conventional twin airplane. The props will block airflow over the wing and the aircraft will experience a large loss of lift. Test it at high altitudes first.

magicrub, 2018-01-14:
@iskess that sounds like wise advice! I can see how that would disrupt your lift so should be used with care. Meanwhile, there are other reasons to allow it safely: 1) turning in-place on a runway; 2) braking hard once you've touched down (common in large manned aircraft). However, option 2 is currently not supported in ArduPilot - motors are totally inhibited once you flare and knowing still-flaring vs touched down is hard to detect without a weight-on-wheels sensor input.

Naterater, 2018-12-01:
After testing conventional reverse-thrust on a twin (believer) recently, there is very little reason to avoid reverse thrust on a twin just because it is a twin. The reverse thrust performed as expected and helped with a significantly steeper approach to landing. Because of the "loss of lift" and increased drag as is expected, the plane naturally loses some elevator authority and gently noses down while reverse thrust is active. With any airplane, reverse thrust can be problematic when engaged in the air due to a reduction in airspeed. Stable planes with the proper CG will naturally nose-down when airspeed is too low in an attempt to recover from their stalled condition.

At 60% reverse thrust, I was achieving descent rates of approximately 6m/s at 18m/s airspeed - thus about 33% slope or 20-degrees on the approach. Very stable and very easy.

I still want to check if this is supported with differential thrust. I guess I might just have to be a guinea pig.

magicrub, 2018-12-09:
Thanks for testing!

Naterater, 2018-12-10:
Tested an autoland today and it was WONDERFUL! Still dialing things in, but it's much better now that my plane doesn't have tons of airspeed after a steep approach.

IamPete1, 2021-02-08:
Looks like reverse with differential thrust implemented, https://github.com/ArduPilot/ardupilot/blob/742d5a9fa00e7599993dcf5ed094db4b8a6ab4fa/ArduPlane/servos.cpp#L695

Feel free to re-open if its broken

closed 2021-02-08 by IamPete1