ArduPilot/ardupilot #7536: Reverse thrust switch on separate channel

Naterater, 2018-01-14:
#### Issue details
Newer ESC's have capabilities to reverse thrust when a high PWM signal is sent through another channel. The ESC will then respond with the same throttle curve (0-100%), but in reverse. This is currently the opposite of reverse thrust as I understand it now.

This new hobbywing manual describes it pretty clearly. [https://cdn.shopify.com/s/files/1/0109/9702/files/HW-SM100DUL_FlyFun-V5_20170307.pdf?14969696128743591372](url)

> "Reverse Brake: After selected this option, the Reverse Brake signal wire (its signal range must be the same as the throttle range) must to be plugged into any vacant channel on the receiver, and you can control the motor direction via that channel. The channel range of 0-50% is the default motor direction, the channel range of 50% to 100% will cause the motor to spin counterclockwise. The channel stick should be within the channel range of 0-50% (0 would be better) when the first time you power on the ESC. After the Reverse function is activated, the motor will stop first and then spin in the reversed direction and then increase to the speed corresponding to the throttle input. Either signal loss, no matter reverse brake signal loss or throttle signal loss during the flight, can cause the throttle signal loss protection to be activated. "

#### Version
3.8.X

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
All traditional planes

#### Hardware type
Pixhawk - all versions

#### Logs

Naterater, 2018-01-16:
This dual channel control could also be applicable in IC engines with variable pitched props. Having two channels to control throttle and prop pitch might be useful. For example as more reverse pitch is added, more throttle is needed. It's slightly off-track from my initial intent, but maybe more useful to others.

simjman, 2018-11-27:
Is this planned to be implemented? I want to use this!

magicrub, 2018-12-10:
I believe @tridge just added it to master.

custeve, 2018-12-10:
Also seconding this request - though it sounds like it's been taken care of. Castle ESCs can reverse only with the use of the aux input of the ESC, main PWM in is throttle percentage, aux PWM is direction with a high signal (1900us) being reverse and low being forward(1100us).

Naterater, 2019-12-10:
> I believe @tridge just added it to master.

I'm pretty sure this was an RC_Options parameter, not a high/low signal OUTPUT when the ESC is supposed to be forward/reversed.

srmainwaring, 2021-05-16:
@Naterater did you establish whether the change supported dual channel ECSs for use with throttle control or was for some other purpose?

These ESCs can be supported using Lua scripting. I've implemented a script for a skid steering rover / boat which flips the reverse channel when the throttle is negative. The script code is a transcription of the rover AP_MotorsUGV::output_skid_steering to Lua with a minor modification for the extra channel. It works but the main issue is that on the ESC's I'm using there is a delay (approx. 1s) between setting the reverse channel high and the motor activating. I'm going to see if I can rework the mixer to account for the delay, but if you are aware if anyone has a solution to this I'd be interested if you have a reference or link.

fredrikfalkman, 2022-07-03:
Hej! I too would like to see this. Would like to use it for automatic landings. Thanks!