ArduPilot/ardupilot #7543: Use AirSpeed Sensor (e.g. MS5525ASO ) as Baro Sensor

liang-tang, 2018-01-16:
#### Issue details

Should we implement  ms5525 driver at AP_Baro library，or Extend AP_Airspeed library？
#### Version
_What version was the issue encountered with_

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_
[image]

DavidIngraham, 2018-01-16:
The Static version of the MS5525 should work directly with the MS5611 driver. I have never tested it because they are not available in low quantities, but the I2C protocol and addressing is identical.

magicrub, 2018-01-16:
Would be nice to confirm it's 100% driver compatible. Likely is, other than a WHOAMI maybe, as @DavidIngraham said. Either way, this is a bad device to use. The cabin pressure changes depending on throttle which makes it a nasty sensor input to fool the control system.

DavidIngraham, 2018-01-16:
I believe the MS5525DSO-FB015AS is actually well suited for use as a barometer. This version of the sensor is an absolute pressure sensor with similar specs to the 5611. The hose barb allows you to connect it to the static port of your pitot static system (you can just tee it), which should give you the best possible static pressure measurement

kd0aij, 2018-01-16:
I agree with @DavidIngraham  that it would be better to have the absolute pressure sensor properly connected to a static port, at least for fixed-wing vehicles. 

The only question is how to provide physical access to the hose barb; the sensor could be a separate I2C module to make it easier to route the static pressure line.

dagar, 2018-01-16:
You also need to set the coefficients.

[image]

magicrub, 2018-01-17:
@dagar If that's the case, I sure hope there's a WHOAMI in there to differentiate between the hardwar versions. Have you worked with the different 5525 flavors and can you confirm?

WickedShell, 2018-01-17:
@magicrub I'd doubt there is. We don't have a WHOAMI to differentiate a MS5611 from a MS5525 let alone sub versions.... People have had problems detecting their extra baro's as airspeed sensors (and vice versa)

dagar, 2018-01-17:
I've only personally handled the mRo version (I wrote the PX4 driver), but there's no WHOAMI.
Next to the CRC byte there's a "serial code", but I'm not actually sure what that is. Let me see if I can get it quickly.

magicrub, 2018-01-17:
oOooO yeah, I forgot. This silly sensor has no WHOAMI and there's different versions that have an offset in the value so you have to guess which device it is depending on if the altitude is above Mt Everest or not. [facepalm].

dagar, 2018-01-17:
I still don't know what the "serial code" is supposed to be, but it's 0 on a random ms5525 I tried.

[image]

I also checked "factory data and the setup" and it's 0x01.

liang-tang, 2018-01-18:
MS5525DSO test result: temperature is so weird, over 400 ℃, and never changed. 

[image]