ArduPilot/ardupilot #7550: AP_Mount: Isn't DataFlash a singleton and _dataflash superflous?

olliw42, 2018-01-16:
I think DataFlash is now a singleton, and accessible as DataFlash_Class::instance(). In my limited understanding I would thus think that the  DataFlash_Class *_dataflash; field in AP_Mount.h is superfluous, and DataFlash_Class::instance()->XXX should be used instead of _dataflash->XXX.

If so, these two lines should be removed:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount.h#L186
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount.cpp#L418

and here _dataflash should be replaced 4 times by DataFlash_Class::instance():
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount_SoloGimbal.cpp#L117-L135

This would be all changes in the AP_Mount library folder.

magicrub, 2018-01-16:
Sounds like you know what needs fixing. Why not create a Pull Request? Would you like some help creating a PR?

peterbarker, 2018-01-22:
On Tue, 16 Jan 2018, olliw42 wrote:

> I think DataFlash is now a singleton, and accessible as DataFlash_Class::instance(). In my limited understanding I would thus think that the
> DataFlash_Class *_dataflash; field in AP_Mount.h is [...]

Yes, that all sounds reasonable!

This isn't the only place in the code where we're passing around objects 
we now have singletons for - the fixes usually make the code much nicer, 
of course!

Did you want to put a PR together for this (with or without help)?

PR #7919 (2018-03-14): AP_Mount: use dataflash singleton
Closes #7550
commit 0652c898 (2018-03-14): AP_Mount: use dataflash singleton

Closes #7550
closed 2018-03-14 by rmackay9