ArduPilot/ardupilot #7554: Rover: allow Steering, Auto using non-GPS navigation

rmackay9, 2018-01-17:
Rover currently requires GPS (not just a good position estimate) to enter Steering, Auto, Guided, RTL and SmartRTL.  We need to modify the "enter_gps_checks" message so that it checks for a good position estimate rather than the state of the GPS.

peterbarker, 2018-01-22:
@rmackay9 See discussion with @OXINARF on https://github.com/ArduPilot/ardupilot/pull/7402

peterbarker, 2018-01-22:
On Wed, 17 Jan 2018, Randy Mackay wrote:

> Rover currently requires GPS (not just a good position estimate) to enter Steering, Auto, Guided, RTL and SmartRTL. We need to modify the
> "enter_gps_checks" message so that it checks for a good position estimate rather than the state of the GPS.

Some things require you to have a position from origin.

Others really do require you to know your GPS coordinates.

Should we add "requires_latlon()" and "requires_ekf_origin_position()" 
("requires_relative_position"?) in place of "requires_gps"?

After all - auto mode strictly requires latlon in our current 
implementation I think.

rmackay9, 2018-01-22:
I think we require the EKF origin to be set before any of these modes will work (we should) so that means that we already know our lat/lon position.

It's true there is a slight advantage to having "requires_relative_position" for Steering and Acro modes.  If that's easy-ish then that sounds like a good thing to add.  It's less critical an issue than not being able to switch into Auto, Guided, RTL, SmartRTL without a GPS though.

rmackay9, 2018-01-24:
This is (mostly) fixed in master now and Rover-3.2.1-rc1.

closed 2018-01-24 by rmackay9