ArduPilot/ardupilot #7557: Passthrough FrSky Telemetry

lthall, 2018-01-17:
Master does not appear to work with Passthrough FrSky Telemetry.

I have tested it against Copter 3.5.4 but upgrading to current master does not work. I have tested with both px4 build and ChibiOS.

floaledm, 2018-01-17:
I'll look into it and report back asap.

lthall, 2018-01-18:
Hey @floaledm, are you from Craft and Theory?

floaledm, 2018-01-18:
Hi Leonard, yes.

floaledm, 2018-01-20:
@lthall Tested the px4-v2 firmware found here (Jan 20) without issue: http://firmware.ardupilot.org/Copter/latest/PX4/

I went from 3.5.4 stable to 3.6-dev.

You could email me the FWs you're using, or suggest another approach to testing...

lthall, 2018-01-20:
Hmmm, ok.

I will have to go back to defaults on 3.6 dev and see if I can get it working.

floaledm, 2018-01-21:
Didn't change the parameters. Just flashed 3.6-dev.

Don't hesitate to email me anything you think might be relevant to test.

DavidIngraham, 2018-04-29:
@lthall Can you confirm if there is still a problem?

lthall, 2018-04-29:
Yes, I am still having an issue but I have not been able to work out why yet. There is something in my settings that stops it working. I need to look into it further.

peterbarker, 2021-07-30:
This works in master; tested it the other day.  I hope Leonard got to the bottom of this eventually :-)

closed 2021-07-30 by peterbarker