ArduPilot/ardupilot #7561: Plane: self touch-n-go for tuning

magicrub, 2018-01-17:
Trigger touch-n-go automatically.

Here's Kevin Garland using [a test branch I made for him](https://github.com/magicrub/ardupilot/commit/bc2393e9cf0b299fb358ee359613cec6c6135369) where he made this [video](https://www.youtube.com/watch?v=xqkmGOor3yk).

magicrub, 2018-01-17:
In an effort to contain the can-of-works known as our param list, it would be nice to create some LAND_ param names that can be reused for a variety of abort schemes. Future isdeas that should be considered:
- abort due to wind
- abort if heading is not withing a given 
- abort altitude (going down)
- abort altitude (climb up to)
- abort bitmask (a bit field of different types of aborts to allow?)
- allow abort from lidar correction too steep (See LAND_ABORT_DEG)
- allow abort via RC throttle input (see LAND_ABORT_THR)
- abort after you land (after you slow down a bit)

Considering those ideas, how about
LAND_ABORT_ALT - depends on LAND_ABORT_TYPE
LAND_ABORT_TYPE - maybe a bitmask? Maybe an enum?
LAND_ABORT_CNT - limit of max self aborts allowed (so we don't self abort forever and never actually land)

rmackay9, 2018-01-19:
I like your ideas about naming.

palm369, 2018-01-20:
Very nice idea Tom!
I would also suggest an option to restart landing on an further away WP than the one before the landing WP. When you set the landing WP very near and low (like you do for very well tuned planes) its not sufficient to just climb to "abort altitude" restart the glide slope. Think of when the airspace is not clear around this WP.  Maybe one can flag a waypoint while creating a misson as the "go to this WP on landing abort"

> LAND_ABORT_CNT - limit of max self aborts allowed (so we don't self abort forever and never actually land)

And whats next? Loiter?

Also would be cool if you could have different landing approaches which the plane selects based on wind direction. Eg: my field has a traditional two strip cross that allows 4 different Landing directions. Based on what wind direction actually is I have to edit the mission before every flight. I here could just program all 4 directions and let the plane decide.

magicrub, 2018-01-20:
@palm369 most of those ideas are discussed and collected here https://github.com/ArduPilot/ardupilot/issues/7055