ArduPilot/ardupilot #7564: Set INITIAL_MODE to GUIDED will send Rover to (0;0) after arming

MikeCharikov, 2018-01-18:
#### Issue details
We can set INITIAL_MODE to GUIDED. When entering guided mode we set the target as the current location:
    set_desired_location(rover.current_loc);

current_loc = 0,0 on startup, cause GPS is not initialized yet.

After arming vehicle will move to zero coordinates. 

P.S. Same behaviour if we set MODE1-MODE6 parameters to GUIDED. 

#### Version
Rover-3.2

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

#### Airframe type
rover, boat

#### Hardware type
SITL

#### Logs
_Please provide a link to any relevant logs that show the issue_

khancyr, 2018-01-18:
Hello, 

Thanks for finding this bug ! We will correct it

rmackay9, 2018-08-22:
Actually I've noticed that the first time a vehicle is switched to Guided mode, if it doesn't have a target it will attempt to drive off to 0,0 (which is off the west coast of Africa).  It's very repeatable in SITL:

1. arm vehicle is any mode (like MANUAL)
2. switch vehicle to GUIDED mode

closed 2018-08-22 by rmackay9