ArduPilot/ardupilot #7565: Why the coverity scan from travis is sending the push to a private email account?

hiro2233, 2018-01-18:
#### Issue details

This is my first issue, you hopefullly doesn't close it at first time please, as ever. Let to the people see it and participate with a democratic mind please.

I have a doubt about coverity scan and travis integration, i followed to the team day by day step by step 24/7 in near about 2 years, and as you see i'm triying to be cooperative with you from at the first time and i'm gratefull with the project too, trying to be free and open sourced human, but, why the coverity scan is sending the pushed PRs to a private email account? so, the data that are building on the automation system is uploading to something there that the people can't acces to this?

A second doubt, why Intel is signing the project? so this is a fork from the original project? the rights was sold, transferred, etc, to Intel?

And a the final doubt, the problem from PX4 and Ardupilot project with the Dronecode was a simple show to the community to skip certain legal mistakes?

I think that answering these doubts, many people could have more clarify their "WAYS" or i'm being imaginary?

Thank you very much for reply.

#### Version
_What version was the issue encountered with_

#### Platform
[X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

hiro2233, 2018-01-18:
and please, be punctual like me, don't answer with your favorite senteces: "what do you mean with blah... blah... blah...?" or "I don't understand what is your point..." or something like tangential comment to try close a incovenient doubt or a development target that like you or needed.

khancyr, 2018-01-18:
Hello, some question are interesting but this page is for reporting confirmed bugs and requesting specific enhancements to the project. Your question is better suited for the ArduPilot forums (https://discuss.ardupilot.org/). I recommend that you post your question over there.

closed 2018-01-18 by khancyr

khancyr, 2018-01-18:
To anwser some questions : 
- autotest build on ardupilot server are accessible here : http://autotest.ardupilot.org/history/
- I think that Tridge mail was used because the infrastructure wasn't ready when coverty was add, it can be changed.
- Intel develop some feature, and share their files with the project, so that normal that they write they made those developpements.
- About PX4 and Ardupilot, I don't undestand your question ?

auturgy, 2018-01-18:
You have asked a number of questions here. As Pierre has asked, please use the forum for questions rather than putting them here, and ensure you provide hyperlinks to the source of your questions, so that the context is clear.
Pending that: no, Intel doesn’t own ArduPilot. ArduPilot is a community project, and a member project of Software in the Public Interest (https://www.spi-inc.org/).
Intel, like many companies, has contributed to ArduPilot development for some years, and some contributed code contains Intel copyright labels. This is normal, appropriate and no different from individual authors doing the same. The License remains extant.
Re personal email addresses: as a community project, many of the developers use personal email addresses to receive ArduPilot correspondence and as login/usernames. This is normal for this project structure and minimises the overhead of having to administer mail servers, and makes life easier on the devs - most of whom are volunteers.
The results of Travis and Coverity scans are not hidden.

hiro2233, 2018-01-18:
@auturgy thanks for reply, i posted it on the forum and linked it.

The problem that you don't know is that all building from PR and forked project too, it's upload the tarball via rsync to Coverity Scan from travis, and it can be accesible to remote state with all privates development from forks, so all projects that are in development mode on personal forks are viewed enterely on the coverity scan and into the travis, for that reason the people that have the admin rights promote to developer to test it first in travis. So the people that have the admin rights can manipulate, extract and copy that to their convenience and then craftily write their rights without a consent from the author developer.  or maybe you are an accomplice too?

hiro2233, 2018-01-18:
I post an example about how do you need to be gratefullness with the people that can be do  possible a great development, and for the in God we trust.

The gratefulness for the functor author is a thing that the Intel people missed, like ever, they are passed over the rights, oviously, the good faith and the best developer around the world maybe forgot that exist people that have a license on the right hand to replace their reputation.

The original author about the Functor is an Australian developer, their name is Don Clugston.

https://github.com/UrusTeam/FunctorDelegateCXX
https://www.codeproject.com/Articles/7150/Member-Function-Pointers-and-the-Fastest-Possible

auturgy, 2018-01-18:
Sigh.
I gave an accurate, polite reply to your questions and now you respond with this nonsense.
I think you may be deliberately trying to pollute the ArduPilot git history with confusion and misinformation.

Please don’t. This project is for developing and promoting control software for autonomous vehicles, not conspiracy theories.

Pedals2Paddles, 2018-01-18:
Now he thinks there a deeply embedded conspiracy in the auto test system.  At first I thought maybe there is just a language / translation thing making it sound like he's accusing ArduPilot of vast conspiracy.  But apparently not.  This is simply absurd.

hiro2233, 2018-01-18:
> Please don’t. This project is for developing and promoting control software for autonomous vehicles, not conspiracy theories.

?
ooh... a tangencial like comment? who mention about conspiracy?

hiro2233, 2018-01-18:
This is a report and focus for profesional educational form purposes for everybody that going on the wrong way.

This is my experience developing with the ardupilot project and that i see along into the code and i'm sharing that exprience, if you are trying to ivalidate this, well thats are other thing, maybe could be turn personal with absurd logic?

@Pedals2Paddles read each post again and you will see who are who.

Pedals2Paddles, 2018-01-18:
Who mentioned conspiracy?  You did. lol.

In the interest of this absurdity not clogging up the ArduPilot git, I'm turning off comments on this. I apparently have that button as a maintainer. If the dev team wants to reopen it, they certainly can.