ArduPilot/ardupilot #7569: Rover: add parameters to default EKF origin on startup

rmackay9, 2018-01-19:
With growing use of non-GPS navigation using [Wheel Encoders](http://ardupilot.org/rover/docs/common-wheel-encoder.html) ([Pozyx,](http://ardupilot.org/rover/docs/common-pozyx.html)) and [visual odometry](http://ardupilot.org/rover/docs/common-zed.html) it would be convenient for some users to have the EKF origin (and thus the initial home location) default to the same place on startup.

Our Pozyx support actually already includes Lat/Lon/Alt parameters so we should move these up a level.  We should probably call the parameters ORIGIN_LAT, ORIGIN_LON, ORIGIN_ALT.  Unforunately they will be stored as floats so their accuracy will be a bit low (10cm ~ 1m) but that can't be helped for now.

Because this would be used by at least Rover and Copter I think we should create a new library, pehaps place it in the AP_AHRS directory.  The class will hold the parameters and it might have a single method, "init" which checks the lat/lon/alt parameters are non-zero and then calls the ahrs.set_origin method.

oneTimePad, 2018-01-24:
@rmackay9  so I noticed a circular dependency issue when I tried this method, or I could be missing something due to the fact that I'm not that adept at C++ :). I might be going about this the wrong way too.

 So if the new library needs to access a reference to ahrs to call ahrs.set_origin, then it needs to include AHRS.h. In addition, for beacon to get a reference to this new library it needs to include the header for this new library, which includes AHRS.h. But AHRS.h includes Beacon's header...so there seems to be a circular dependency. I don't see a way around this though...

rmackay9, 2018-01-26:
Hi @oneTimePad, I've thought about this a bit more and this method perhaps removes the circular dependency.

I think we should remove the AP_Beacon::get_origin method and the bit in AP_NavEKF2 (and probably AP_NavEKF3) that calls beacon->get_origin.  It looks like there are two other operations done at the same time, "alignMagStateDeclination()" and "ekfOriginHgtVar = xxx" so these should be removed as well and we can ask @priseborough if he agrees this is OK.

Then I think we should rename Rover::update_home_from_EKF() to Rover::update_home_and_ekf_origin() and at the bottom of that function we should have a call to something like "ekforigin_defaulter.set_ekf_origin_to_default(ahrs)" where "ekforigin_defaulter" (perhaps an odd name but it's obvious what it's purpose is) is an instance of this new class.

Inside this "set_ekf_origin_to_default" method it would call ahrs.set_ekf_origin() if the parameters are non-zero.  This method might also want to first check that the gps.num_sensors() is zero.  I think the GPS object is a singleton now so perhaps it can be accessed from anywhere using GPS::num_sensors().. I'm not totally sure.  If not it could be passed into the set_ekf_origin_to_default() call.  It could perhaps also check that the ekf-origin hasn't already been set by calling ahrs.get_origin() and checking that it returns false.

I think longer term we may want to move the Rover::set_ekf_origin(const Location &loc) so that it goes through this new class but we don't need to do that immediately.

By the way, the other possible way we could do this is by putting these new origin parameters and this new "set_ekf_origin_to_default" directly into the AP_AHRS class.  I'd like to have @tridge's opinion on this.  I'm tending towards having a separate class because the AP_AHRS library is already quite complex.

oneTimePad, 2018-02-06:
Any updates on which method is preferred (new class or putting the params in AP_AHRS)?

IamPete1, 2021-09-12:
This is possible with scripting, I wonder if we need params too? https://github.com/ArduPilot/ardupilot/pull/17907

Of course you could use the script to read in the `SCR_USERx` params...

rmackay9, 2026-05-04:
This has been resolved by AndyP's PR https://github.com/ArduPilot/ardupilot/pull/31177