ArduPilot/ardupilot #7579: apm_sitl.launch

jianhuaixie, 2018-01-21:
while processing /home/xxx/catkin_ws/src/mavros/mavros/launch/apm_sitl.launch:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/jianhuaixie/catkin_ws/src/mavros/mavros/launch/apm_sitl.launch'﻿

khancyr, 2018-01-21:
Hello,
This file doesn't exist nor here nor in mavros. You need to create one.
What instructions were you following ?

jianhuaixie, 2018-01-21:
@khancyr I following PX4. I create one the content is just: python apm_sitl.py?

jianhuaixie, 2018-01-21:
https://github.com/erlerobot/mavros/blob/master/mavros/launch/apm_sitl.launch

khancyr, 2018-01-24:
that is just an xml file. just create it with your favorite text editor. I advice you to look at ROS tutorial on how to use launch file !
This place is for code issue, please use https://discuss.ardupilot.org/ for help.

closed 2018-01-24 by khancyr