ArduPilot/ardupilot #7581: Copter: remove battery failsafe exeption for Auto mode

rmackay9, 2018-01-22:
Copter's battery failsafe feature has an exception for Auto mode.  Even if the user has set the FS_BATT_ENABLE parameter to "Land", if the vehicle is in Auto, it will always RTL.

https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/events.cpp#L65

I can't imagine why this exception is required.  I think it's more likely that a user would be confused why the vehicle does not do what it has apparently been configured to do.

I suspect this exception is a relic from a time when Copter's failsafe response could not be configured for each failsafe types (i.e. rc/transmitter and battery failsafes responses were combined in a single parameter).  Certainly for the RC/Transmitter failsafe we need special handling while in Auto mode because users sometimes want to configure the vehicle to continue with the mission because they plan for it to fly outside of RC/transmitter range.  This is not the case for the battery failsafe however.

Pedals2Paddles, 2018-01-22:
I saw this a few weeks ago working on the bitmask options and was a little perplexed by it. I can see wanting it to RTL if it's off on a mission. But it does indeed do something that isn't what the operator selected as an option, and isn't published or expected

rmackay9, 2018-02-05:
Fixed in master and wil go out with Copter-3.6

closed 2018-02-05 by rmackay9