ArduPilot/ardupilot #7600: ArduPlane: make quadplane weathervaning much smoother

expntly, 2018-01-24:
#### Issue details
Weathervaning is a quite aggressive and harsh on the vehicle as it is today.  I turned it on on a vehicle, with low WVANE and VFWD gains, and I had to bring the vehicle down soon after take-off.  I also tried an AUTO mission with it and as it was going downwind on the last leg before VTOL_LAND, a lot of things happened at the same time while de-transitioning.  At a velocity close to cruise speed it 1) yawed aggressively 180 degrees to turn into the wind, 2) it banked ~60 degrees, 3) it kept going towards the VTOL_LAND waypoint while drifting/losing speed (as it should) except backward.  That's a lot to handle for a quadplane.

I ran a simulation just to make sure and this yielded the same results, see below.

What I think might be better is to decouple the transitions from the weathervaning phase.

* before VTOL to cruise transition or after cruise to VTOL transition is complete: no change, vehicle weathervanes into the wind, cool

* when VTOL -> cruise transition starts: first yaw so its heading is directly towards the next waypoint, then turn on horizontal propulsion (this might actually be the default regardless of weathervaning);  it should be better than what it does today which is basically everything all at once and leads to 1) a vehicle going where it's "not supposed to" (x), 2) yaw+roll and turn when transitioning, which isn't great

* when cruise -> VTOL detransition starts: let the vehicle de-transition as if weathervaning were turned off, then when the VTOL_LAND waypoint is reached, yaw into the wind and start descending

x : say the wind is coming from some installation or building, the plane will head towards these and make a turn all while trying to transition -- super risky / unsafe.

A simple way to fix this is maybe to 1) dampen the weathervaning gain just before transition/detransition and 2) make the vehicle yaw towards the next WP before transition to cruise.

#### Version
3.8.2

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
QuadPlane

#### Hardware type
PH2.1

#### Logs
https://drive.google.com/open?id=15oQ9noYn-tfZEVWv-U_WL8eVI0-Ogvw0

expntly, 2018-01-31:
Any of the devs had a chance to look into this?  Or perhaps I'm doing something wrong with weathervaning?  It just seems unusable today and actually very risky to enable it.

DavidIngraham, 2018-02-07:
Hi @expntly,
The Q_WVANE_GAIN in your simulated log is set to 0.5. Is there a reason that you chose to use such a high value? The documentation recommends beginning with a value of 0.1 and increasing up to 0.4 for a "rapid response". I think this value may be set too high.

However, I do think it will be good to discuss your latter two bullet points. We may want to split this issue into two to address those. This would give us the following two issues:

**1. Yaw towards waypoint as VTOL before Auto transitions**
I'm not sure that this is always the correct behavior. If there is any significant wind, it is safer to transition directly into the wind rather than trying to transition with a crosswind. However, the current behavior will attempt to turn towards the waypoint immediately upon transition, which is also not correct.

I see two valid options here, depending on operator preferences:
    - Disable wind veining entirely, and rely on the operator choosing an appropriate upwind location for the initial waypoint (This is possible now)
    - Have the vehicle windvane, but change the behavior such that the vehicle locks onto a heading and transitions towards a projected waypoint until the transition is complete, then fly to the initial waypoint as a fixed wing. (This will require a modification to the transition logic)

The above two options are both valid, but operations requiring higher levels of automation will require the latter. However, if there are obstacles in the operational area that must be avoided during transition, only the former can be used.

**2. Disable WVANE during the Q_POSITION1 phase of FW->VTOL Transitions.**
I'd like to get @tridge's input on this if possible. This seems like the appropriate action to me.

jace26, 2018-05-30:
This has been up here a while it seems but I'm in the same boat with this. Detransitioning a large quadplane with weathervaning on is a dangerous proposition as is, however with a large bird this function is almost essential in VTOL because our wing area is enormous and has a lot of inertia. 

I think a much simpler way to handle this is to limit weathervaning by fwd velocity, and also by altitude. In some cases I don't want the aircraft yawing into the wind at 2m when our wings are this long. 

I'm on the outside looking in here, but glancing at the code it seems that we need 2 new parameters: Q_WVANE_VELMAX - the velocity below which weathervaning will occur
Q_WVANE_MINALT - the altitude above which weathervaining will occur

 Adding these into the limiting conditions of the get_weathervane_yaw_rate_cds starting on line 2366 of the quadplane.cpp looks like it would fix this problem.

@tridge  was the last to edit this section and I'd like his opinion.

IamPete1, 2022-08-05:
Many quadplane improvements since this issue was opened. We now have a min weathervaning height and max speed. Also a option to disable for takeoffs so you transition in the direction the vehicle is setup in. https://github.com/ArduPilot/ardupilot/pull/19859

closed 2022-08-05 by IamPete1