ArduPilot/ardupilot #7604: Copter: move get_pilot_desired_lean_angles to mode.cpp

rmackay9, 2018-01-24:
There are a number of methods in Copter's Mode class that simply call through to identical methods in the Copter class.  We should move as many of these as possible directly into the mode class.

https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/mode.h#L95

This is a non-functional change, it just makes the architecture a bit better.

peterbarker, 2018-01-24:
This is an example of a commit making such a move:

https://github.com/ArduPilot/ardupilot/pull/7458/commits/cd51b6a8696d50bbc4eb4147915eb6f6b4e31b80

It's part of a larger cleanup - and many of these functions may have similar cleanups around them.

starlord1311, 2018-02-12:
can i take up this issue? @rmackay9

rmackay9, 2018-02-12:
@starlord1311, yes you can!  go for it!

starlord1311, 2018-02-13:
can you help me with what i am supposed to do?

rmackay9, 2018-03-14:
Fixed by @EbinPhilip, thanks!

closed 2018-03-14 by rmackay9

peterbarker, 2018-03-14:
@EbinPhilip If you're in a mind to clean this up some more...

The first two arguments aren't required; every caller supplies the same ones!

There are also a pair of outdated comments in there about returning floats.  Pretty sure this does that now :-)

EbinPhilip, 2018-03-14:
@peterbarker I'm sorry, but i didn't really get what you said. 
You want the roll_in and pitch_in variables removed as arguments and made part of the mode class? I can do that. Which comments again?

peterbarker, 2018-03-14:
On Wed, 14 Mar 2018, Ebin Philip wrote:

> You want the roll_in and pitch_in variables removed as arguments and made part of the mode class? I can do that.

Yep.  They can be filled in by the function itself.

> Which comments again?

https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/mode_stabilize.cpp#L46

That's not the only instance :-)

EbinPhilip, 2018-03-16:
Ok got it. I'll make the changes. Since the function has to return two values(pitch and roll), I'll define a new structure to hold both in Copter.h. That should do right?

peterbarker, 2018-03-16:
On Thu, 15 Mar 2018, Ebin Philip wrote:

> Ok got it. I'll make the changes. Since the function has to return two
> values(pitch and roll), I'll define a new structure to hold both in
> Copter.h. That should do right?

I wasn't suggesting changing the method of return.  Just not passing in 
parameters that the mode knows all about anyway.

EbinPhilip, 2018-03-18:
Did that. PR #7954