ArduPilot/ardupilot #7616: Rover: add support for L298N is a dual H-Bridge

rmackay9, 2018-01-26:
The L298N H-bridge requires two pins to control the direction of each motor.
http://howtomechatronics.com/tutorials/arduino/arduino-dc-motor-control-tutorial-l298n-pwm-h-bridge/

I'm not sure how common this motor driver is but we could add support for this by using an extra relay pin for each motor.  The output for the extra relay pins could probably always just be the inverse of the existing pins although it's possible we might want to set both pins to the same value if we know we want the motor to stop (i.e. the vehicle is disarmed or input throttle is zero).

hiro2233, 2018-01-26:
it's very common, and it's the base driver motor for all common 3D printers like mendel model and with sanguinololu boards.

khancyr, 2018-01-26:
the motor brake feature isn't implemented thus.

xlla, 2019-05-10:
TB6612 need two relay_pin too

vnsiot, 2020-06-01:
Hello,
Can you give the wiring diagram for connecting the L298N Dual H-Bridge motor driver to Pixhawk?

Ebaker310, 2020-07-11:
@khancyr How do I configure the relay pins to use the L298N motor driver?

See my comment on this thread: https://discuss.ardupilot.org/t/radiolink-pixhawk-ardurover-and-l298n-h-bridge/25228/10?u=ebaker310

rigid, 2020-10-31:
I just encountered the same issue and I'd appreciate if those kind of ESCs would be supported. 

Those kind of H-Bridge drivers are quite common. They are basically "Normal (PWM)" type using 2 PWM output signals. One for forward and one for reverse. (Both outputs should not be high at the same time).

heeplr, 2020-11-13:
I'm slightly confused. While I see why @khancyr says that these kind of controllers are already supported, I still don't get it to run since SERVOx_MIN output is limited to 800 as minimum, but those controllers need 0 as minimum. With 800, the motor is spinning at 0% throttle.

Am I doing something wrong?

khancyr, 2020-11-13:
@heeplr , hello,

You need to set the PWM_TYPE to BrushedWithRelay ! https://ardupilot.org/rover/docs/parameters.html#mot-pwm-type-motor-output-pwm-type

heeplr, 2020-11-13:
I did that (and rebooted) and still SERVO1_MAX has a valid range of 500 to 2200 (not 800 as written above).
I tried rover v3.5.2 and 4.0.0.

khancyr, 2020-11-13:
on PWM the output is a real pwm not a rcpwm, so we don't care about SERVOx_MIN/MAX. Those are hard setted to 1000 - 2000 for easier calculation and do 0 - 100% duty cycle mapping !

heeplr, 2020-11-14:
ok, i'm one step closer.
@khancyr thanks a lot for clarifying and thus saving me a ton of time troubleshooting.

After I did a factory reset, I get 0-100% duty cycle mapping. (The 0% throttle position is very very small, though. So the Motor only stops spinning when the throttle is exactly at the right position and even slight touches starts running it.)

But now i'm struggling with relays. There are two Problems:
* according to http://ardupilot.org/rover/docs/common-brushed-motors.html the Relay can be assigned to an AUX Pin, but it only works with my Pixhawk 2.4.8 when I use AUX Pin 6 and AUX Pin 5. Is this an issue of my hardware or an issue with the wiki?
* Most importantly: There seems to be only one relay supported but H-Bridges need 2 Pins:
   **Forward**: Relay 1 must be HIGH, Relay 2 must be LOW
   **Reverse**: Relay 1 must be LOW, Relay 2 must be HIGH

I tried the remaining Relay Pins, but they don't seem to change. Is this already supported by ardupilot and I'm doing it wrong (again)? If it isn't, what would be a good way to implement it? (I found https://github.com/ArduPilot/ardupilot/blob/master/Rover/AP_MotorsUGV.cpp but i'm not sure what's exactly needed to cleanly implement a second relay pin that's an inverted version of the first one.)

juvinski, 2020-11-14:
Hi @heeplr 
if you are using rover, you can take a look on this branch - https://github.com/juvinski/ardupilot/tree/RoverBridgeDualRelayPins
I create a DualRelaypins and the logic of the outputs are working as expected.

heeplr, 2020-11-17:
@juvinski Cool, thank you! Your _BrushedWithTwoRelays_ solution would be exactly what's needed.

I'm having trouble testing it, tho. (Something seems wrong with my build environment since compiled fmuv3 images don't boot - no tone, status LED stays off after FMU B/E LED stops blinking. The downloaded images work fine.)

I'll keep trying and report my findings here.

juvinski, 2020-11-17:
hi @heeplr ,

what kernel version are you using ?

heeplr, 2020-11-17:
you mean ChibiOS version or linux kernel? how to find out?
i tried most recent master (recursive clone) and older versions. I gathered some build output of an older version here: https://gist.github.com/heeplr/5e34b64db7c73d4f6c9f7b5a12b53920 ... it always ends up with the same error.

juvinski, 2020-11-17:
Hi @heeplr 
I was thinking you was using beaglebone. Sorry but chibios I don't know.

heeplr, 2020-11-17:
@juvinski thanks anyway

heeplr, 2020-11-21:
@juvinski I got it running. Works like a charm. (I'm still struggling with conflicting AUX pins when using wheel encoders but that's another issue.)

Any chance for this to be merged into master as a pull request?

juvinski, 2020-11-22:
Hi @heeplr 

I create a or for that but after some discussions https://github.com/ArduPilot/ardupilot/pull/14007

corteXnzt, 2021-03-15:
I want to use L298N or TB6612 but come across this issue. Trying to build the RoverBridgeDualRelayPins branch and it shows this error. 

`Build failed
 -> task in 'objs/AP_NavEKF2' failed (exit status 1): `
 
 I solve it by replacing the files in this pull request. I'm running Ubuntu 20.04. Commenting it here in someone encounters this error when building the RoverBridgeDualRelayPins branch.

https://github.com/ArduPilot/ardupilot/pull/13921

Hope master branch will support drivers like L298N /TB6612 with 1 PWM speed control and 2 pins for direction control for each motor.

I have other motor driver but requires 2 PWM on each motor but is also not supported I think.
https://www.cytron.io/p-3amp-4v-16v-dc-motor-driver-2-channels

corteXnzt, 2021-03-15:
> Hi @heeplr
> if you are using rover, you can take a look on this branch - https://github.com/juvinski/ardupilot/tree/RoverBridgeDualRelayPins
> I create a DualRelaypins and the logic of the outputs are working as expected.
> 
> BTW, you can use the RCX_DZ to set a dead zone bigger and the middle point will be more easy to control.

@juvinski 
I'm trying to make this work but there is no response on both motors

uploaded manually built firmware from your branch
set servo output 1 to steering
set servo output 3 to throttle
set  config>basic tuning>BrushedWithRelay

set relay pins
[image]

How will I set this pin 3-6 as controls for direction of L298N? Or automatically configured? Do you have some wiring diagram of this? I could not find connection diagram even in the forums and they have the same problem.

DroneMang, 2021-05-21:
Is https://github.com/juvinski/ardupilot/tree/RoverBridgeDualRelayPins supported yet in master or the beta firmwares?  I want to use the L298N driver with rover and 2 12v motors.  Thanks!

rmackay9, 2022-05-31:
We seem to have support for this motor now but we should add @jazzl0ver's instructions to the wiki so that people can find them more easily ([raised issue here](https://github.com/ArduPilot/ardupilot_wiki/issues/4375))

I'll close this issue now and thanks very much to @jazzl0ver!

closed 2022-05-31 by rmackay9

jazzl0ver, 2022-05-31:
Thank you, guys, for such a great product and support!