ArduPilot/ardupilot #7619: Copter: battery failsafe logged even when FS_BATT_ENABLE = 0

rmackay9, 2018-01-29:
It is possible (as reported by Iketh in [this discussion](https://discuss.ardupilot.org/t/copter-3-5-5-rc1-available-for-beta-testing/25162/12)) for the a battery failsafe ERR message (Error 6) to be recorded to the dataflash logs even when the FS_BATT_ENABLE parameter is zero.  Within the code, the failsafe.battery flag is also set to true.

This is different from how other failsafes are handled.

There are perhaps two options:

1. change the FS_BATT_ENABLE parameter description so that "0" is "Report Only".  This at least makes it clear to the user that it's not possible to disable the failsafe.
2. add a new option "5: Report Only" which is the equivalent of the current behaviour and change the parameter default from "0" to "5" (this will mean most users who have not set the parameter will see no change in behaviour).  Change the "failsafe_battery_event" method so it exits immediately if the parameter is set to "0" (or perhaps if "0" it should also clear the battery failsafe flag if set).

https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/sensors.cpp#L207

Pedals2Paddles, 2018-01-29:
Isn't this correct behavior? The wiki explains that it will still generate the low battery alert messages and tone alarm if you have voltage and mAh values set. I think all it needs is to not send an error with the word failsafe in it.

rmackay9, 2018-01-29:
It's at least different from other failsafes.  For example with the RC failsafe, if it is disabled, there's no reporting at all.  With ADSB_Avoidance failsafe, we have separate disabled and report-only options.

Pedals2Paddles, 2018-01-29:
Indeed. But it is also doing exactly what the documentation says it will do.  http://ardupilot.org/copter/docs/failsafe-battery.html

If anything, it should probably just not use the word failsafe in the messaging?

drjackso, 2018-01-30:
There is an additional bug, where if FS battery low is triggered Pixhawk will not check the voltage next time it is armed. i.e. if FS_BATT_ENABLE and set true, then battery is charged without resetting Pixhawk, Pixhawk will not arm due to "Check Battery".

Pedals2Paddles, 2018-01-31:
> There is an additional bug, where if FS battery low is triggered Pixhawk will not check the voltage next time it is armed. i.e. if FS_BATT_ENABLE and set true, then battery is charged without resetting Pixhawk, Pixhawk will not arm due to "Check Battery".

I don't think that's a bug, it's intentional. But I understand that it does get in the way i you swap flight batteries without powering down the flight controller.  But that's probably pretty rare for someone to do. And it's also the only way to reset the battery to 100% MAH.

Pedals2Paddles, 2018-02-01:
This should take care of the problem. https://github.com/ArduPilot/ardupilot/pull/7639

rmackay9, 2018-03-28:
So basically the user can't turn off the warning unless they set the FS_BATT_VOLTAGE (or whatver it's called now) to a low number.   I think this is OK.  We don't get a lot of complaints about it... we could just "handle it with documentation" by adding something to the wiki.

WickedShell, 2018-03-28:
Alright, closing is as intended behavior then.

closed 2018-03-28 by WickedShell