ArduPilot/ardupilot #7620: Copter: sanity check FLTMODE_CH parameter

rmackay9, 2018-01-29:
We should add a check that the FLTMODE_CH parameter is a reasonable value (i.e. from 1 ~12) to ensure we don't try to read beyond an array boundary (or something like that).

It's possible that @peterbarker's PR to add RCx_OPTION may resolve this item.

This issue was raised by @olliw42.

tridge, 2018-01-29:
why is a value above 12 invalid? many receivers have 16 channels
The hal.rcin API already does bounds checking I think

rmackay9, 2018-01-29:
It looks like RC_Channels::rc_channel(chan) returns a nullptr if the channel provide is out of range so I suspect something bad would happen.

>     static RC_Channel *rc_channel(uint8_t chan) {
>         return (chan < NUM_RC_CHANNELS)?&channels[chan]:nullptr;
>     }

WickedShell, 2019-04-05:
All consumers of `RC_Channels::flight_mode_channel()`currently handle a null pointer correctly, and wrapping to an arbitrary input channel rather then returning null pointer is more dangerous.

closed 2019-04-05 by WickedShell