ArduPilot/ardupilot #7625: Rover: support balance bots

rmackay9, 2018-01-30:
We should add support for Balance Bots ([description from make magazine](https://makezine.com/projects/arduroller-self-balancing-robot/)).  This will likely involve:

- extending the AR_AttitudeControl library to add a pitch angle to throttle/acceleration control.
- ensure all existing drive modes work with balance bots.
- document the setup on our wiki.

jmachuca77, 2018-01-30:
Cool video, I had read the article but I had not seen the video. I wonder if Jason’s code could be a starting point for this

rmackay9, 2018-01-30:
Jason's code will be very old now but if we could find it, it's probably a very good place to start.  It would have the basic control stuff in there.  @tridge has found this repo from @jschall which has some balance-bot code it in as well: https://github.com/jschall/ardupilot/tree/balancebot

auturgy, 2018-01-30:
Jason’s code is here: https://github.com/jason4short/ardupilot/tree/ArduRoller

Regards,

James

jschall, 2018-02-01:
If there's any interest in hardware for this, let me know. I have designed a 2m tall direct-drive brushless balance bot. They're pretty expensive and time consuming to print and build, but I would be willing to provide one to someone who wants to make it work with APM:Rover.

Linjieqiang, 2018-03-08:
Nice.

mundsen, 2018-04-13:
Hi, I have started planning a "ArduRoller 32" build (using PixHawk/PixRacer not the old controller) inspired by the old ArduRoller project.
I have ordered 120mm rubber wheels, but find it hard to decide if I should base the bot on Nema17 or 6V / 12V Geared DC motors with encoders like in the original ArduRoller??

Hope the PixHawk get balance bot support soon.

Great if anyone can held me decide so I can order the parts..

rmackay9, 2018-07-06:
Support for balance bots isn't perfect yet but the basics are in and will be released with Rover-3.5.  Future improvements to the balance bot code can be handled as new issues.  Thanks @EbinPhilip!

closed 2018-07-06 by rmackay9