ArduPilot/ardupilot #7627: Problem in running Ardupilot with jsbsim

padmashree9110, 2018-01-30:
#### Issue details
I am trying to do SITL setup for Ardupilot in UBUNTU14.04.
When I run below command

sim_vehicle.py -f jsbsim:MyModel --console --map

It is not getting connected to the vehicle

#### Logs
https://gist.github.com/padmashree9110/262f8eebac6d7d5c5b99dc4c8f8b5d41

Can anybody suggest me the solution.

Thank You

yahyaelabiad, 2018-03-07:
Hi,
please did you slove it ??, i'am facing the same problem

peterbarker, 2018-03-08:
Check the output in the terminal window that the actual ArduPlane binary is running in.  It might be minimised.

When I run your commandline I get this error in that window:

```
Unable to create jsbsim reset script /home/pbarker/rc/ardupilot/Tools/autotest/aircraft/MyModel/reset.xml
```

Simple fix would be to create the relevant directory.  More complicated fix would be to have `sim_vehicle.py` magically create that directory.  That would be a bit interesting to do in a nice way.