ArduPilot/ardupilot #7629: Copter: EKF goes crazy and loses the attitude, with inevitable crash

robustini, 2018-01-30:
#### Issue details
I've a really bad crash yestarday with "Ardupilot: Copter V3.5.5 RC1" during a "Simple Mode" demonstration, performing continuous rotations of yaw as I often show in my videos.
But imho this time EKF has failed.
The quad started to take a direction increasing the angle more and more, and from the log instead the EKF show the electronics leveled while instead the copter is tilting and acquiring speeds at over 15 m/s, so I completely lost control and the possibility of recovering it.
During this maneuver i see a lot of EKF errors of Position and Velocity, compass switch, it seems that the EKF goes crazy and totally loses the attitude, not understand the real level of the electronics (Pixhawk 2.1).
All the compass are calibrated before takeoff, I see Z mag values in the log that strangely arrive at absurd limits.
Really dangerous, need investigation.
Here the moment when the quad is tilted in flyaway at over 40° angle, really tilted, see the hud horizont, for the EKF is levelled:

[image]

#### Version
Ardupilot: Copter V3.5.5 RC1

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Quad

#### Hardware type
Pixhawk 2.1, Here+ RTK (primary) and Here GPS (secondary)

#### Logs

https://drive.google.com/drive/folders/1DZVZhTtlN7vptDb-kv8xQMEt-Ds2DVeo

priseborough, 2018-01-30:
This is  a known vulnerability of the EKF2 formulation. For rotating airframes (which is what you have effectively created with spinning continuously in simple mode), I recommend you use EKF3 which handles that type of motion better or reduce the max spin rate. I could add support for spinning airframes by 'de-spinning' the sensor data before use in the EKF, possibly using an approach similar to https://discuss.ardupilot.org/t/gyropter-monocopter-support-with-the-help-of-a-fourier-based-transform or by adding another state, but that will require significant effort for what is a niche use case.

priseborough, 2018-01-30:
Also note that the max sustained yaw rate that can be tolerated also depends on the offset of the IMU from the spin axis and the GPS offset from the IMU and whether the IMU and GPS position offset parameters have been set.

night-ghost, 2018-01-30:
> I could add support for spinning airframes by 'de-spinning' the sensor data before use in the EKF

It woud be great for failsafe in case of motor loss [as here](https://www.youtube.com/watch?v=bsHryqnvyYA)

R-Lefebvre, 2018-01-30:
I'm pretty sure that spin rate would saturate the gyros, and then it's done.  Most of that work was done in a Vicon lab with ~1000Hz external pose estimates?  

Also never seen this demonstrated with larger systems (ie: can motor response rate of 12+" propellers keep up with stabilize something spinning that fast)

All interesting ideas of course.  But every time I see somebody point to that I have to bring up the reality.

night-ghost, 2018-01-30:
>But every time I see somebody point to that I have to bring up the reality.

this allows ideas to be born!

>Most of that work was done in a Vicon lab with ~1000Hz external pose estimates?

if to believe the text below the video, this is a usual pixhawk and proprietary software. So it can be done by Ardupilot team too

IamPete1, 2021-08-15:
sounds like the solution is to use EKF3, it is the default now in any-case.

closed 2021-08-15 by IamPete1