ArduPilot/ardupilot #7631: Too complicated to prepare the environment for Building ArduPilot for APM2.x on Windows

kviacheslav, 2018-01-30:
#### Issue details
Toolchain requires to specify the path to the ARDUINO in the config.mk, without it does not work:
$ make -n apm2
../mk/find_arduino.mk:45: *** ERROR: Cannot find Arduino on this system, please
specify on the commandline with ARDUINO=<path> or on the config.mk file.  Stop.

Although in reality this setting does not use, and looking for the path to avr-g++.exe in the PATH, as prescribed in find_tools.mk:
ifeq ($(findstring MINGW, $(SYSTYPE)),MINGW)
  TOOLPATH :=  $(subst :, ,$(PATH))
  FIND_TOOL    =  $(firstword $(wildcard $(addsuffix /$(1).exe,$(TOOLPATH))))
endif
#### Version
https://github.com/ArduPilot/ardupilot/tree/master-AVR

#### Platform
Any
#### Airframe type
Any
#### Hardware type
APM

rmackay9, 2018-01-30:
We don't support the arduino builds any more I'm afraid.

auturgy, 2018-07-30:
Closing. As Randy said, we don’t support that hardware anymore (and haven’t for several years, I’ll add).

closed 2018-07-30 by auturgy