ArduPilot/ardupilot #7632: SITL No module named cv2

tkoch96, 2018-01-30:
I am using python2.7 with the latest versions of MAVProxy. 

I am trying to use SITL with MAVProxy, following the tutorial here http://ardupilot.org/dev/docs/copter-sitl-mavproxy-tutorial.html.

I successfully executed: sim_vehicle.py -w.

I attempted to execute sim_vehicle.py -j4 --map --console, and noticed the error:
 ERROR in command ['load', 'map']: 'module' object has no attribute 'init'

While in the MAVProxy console, I ran 'set moddebug 3' and then ran 'module load map', to which I received the following error.

STABILIZE> Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 281, in load_module
    module = m.init(mpstate)
  File "/Library/Python/2.7/site-packages/MAVProxy/modules/mavproxy_map/__init__.py", line 629, in init
    return MapModule(mpstate)
  File "/Library/Python/2.7/site-packages/MAVProxy/modules/mavproxy_map/__init__.py", line 59, in __init__
    from MAVProxy.modules.mavproxy_map import mp_slipmap
  File "/Library/Python/2.7/site-packages/MAVProxy/modules/mavproxy_map/mp_slipmap.py", line 13, in <module>
    import cv2
ImportError: No module named cv2

ERROR in command ['load', 'map']: 'module' object has no attribute 'init'
Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 495, in process_stdin
    fn(args[1:])
  File "/usr/local/bin/mavproxy.py", line 323, in cmd_module
    load_module(args[1])
  File "/usr/local/bin/mavproxy.py", line 281, in load_module
    module = m.init(mpstate)
AttributeError: 'module' object has no attribute 'init'

However, I am pretty sure I have cv2 installed. For example, on a command line when I type python to open up the (2.7) interpreter and type
import cv2

it imports successfully. 

I'd appreciate any help in resolving this issue, although please be aware this only seems to prevent me from using the map feature.

auturgy, 2018-01-30:
Are you using Windows by chance?

Regards,

James

khancyr, 2019-05-31:
This should have been fixed on recent version of mavproxy

closed 2019-05-31 by khancyr